opentx/opentx #7842: [Opentx] check box not visible for Italian language in HARDWARE menu

OpenMik, 2020-07-25:
OpenTX 2.3.9
Radio X9E Italian language
In HARDWARE menù check box for "Check RTC voltage" is not visible because the italian text is too long
("Controllo voltaggio RTC")
Text may be only
"Controllo RTC"

3djc, 2020-07-25:
Or "voltaggio RTC" ?

3djc, 2020-07-25:
"Controllo RTC batt" or something ?

3djc, 2020-07-25:
I guess actually the best option is to push the items a bit on the right, no reasons to squeeze those

OpenMik, 2020-07-25:
may be a solution push the items a bit on the right
but for this in debug option is necessary to move the [Switches] text under the [Analog ] text
there is no more space on the right

please check also the bluetooth option
first show "---" 
second show "Enabled" (In italian "Attivo")
third show blank space (in italian "Non in mo")

PR #7843 (2020-07-31): Radio hardware screen improvements
- move some items in 212 radio right
- fix X9E BT mode issue

This fixes #7842
closed 2020-07-31 by bsongis