opentx/opentx #7844: [2.3.9] SH switch not in model logfile

0crap, 2020-07-25:
The SH switch which I use as a (dis)arm switch is never logged in my model logfile on the Taranis QX7S.

Other switches I use are logged into the model logfile by default.
Is this a bug or am I misunderstandig something? (Coul very well be the case.)

[image]

**General information**
 - OpenTX version: 2.3.9
 - Hardware: Frsky Taranis QX7S
 - OpenTX Companion: 2.3.9 Windows

3djc, 2020-08-11:
SH is a toggle, so when you activate, it might be ON for a very short period, and therefore not appear in logs as ON

0crap, 2020-08-11:
That would be a bad decision, to have a toggle switch to disarm. :-)
So that is not the case. Might be I swapped it in the past.

[image]

3djc, 2020-08-11:
Cannot replicate

[image]

[image]

Both are properly logged. Do you happen to enable logging on your arm switch too ?

0crap, 2020-08-11:
Hmmm, I'm only aware I could enable/disable logging on found telemetry sensors.
Where can I enable/disable this on switches?

3djc, 2020-08-11:
What i meant is logging is not by default, it needs to be enabled by a sd logs special function. This special function is triggered by a switch. Is it SH or a related LS ?

0crap, 2020-08-11:
OK, I understand.
Yes, it is triggered by my arm switch. 🥇 
Guess that's why.
Sorry for wasting your time.

I played with delay but that only seems to interact when starting logging.
Any smart thing to do, to keep logging a second after disarm? (Not having to use a separate switch.)

3djc, 2020-08-11:
You need to create a LS for that

closed 2020-08-11 by 3djc

0crap, 2020-08-11:
@3djc I tried, using the Delay parameter would be the most logical choice to use.
However Delay is only active on activation. Deactivation happens immediately. (Maybe that is a bug?)

Can you give an example how to keep logging active for a few seconds after disarming?

kilrah, 2020-08-11:
Set the source to be the opposite of what you want, then select the !LS to start logging.

0crap, 2020-08-12:
@kilrah Thx for stepping in.
What you are effectively creating with that is having the Delay function during deactivation.
So yes, that works if you're only starting and stopping logging with the arm switch.
I have more logic depending on the arm switch, so now that doesn't work as intended.

To me it seems a bug that the Delay parameter is only active on activation.
If Delay is also active on deactivation (what you would expect it to do in the first place) would make for a clean solution.

Is Delay only working on activation intended behaviour?

kilrah, 2020-08-12:
Whether intentional or not it's hpw it's been for many years so it's not like it could be changed on a whim. But nothing prevents you from making separate LSes for the various functions.