opentx/opentx #7854: Taranis qx7 display issues after upgrading to OpenTX v2.3.9

fayazmd83, 2020-07-30:
Dear Developer,  

Hope you're doing healthy and safe!  

I recently purchased new Taranis QX7 Access which came with FCC 2.2 firmware  I believe.   I then upgraded the bootloader and firmware to v2.3.9 from the website.  After doing this I started noticing display issues.  please check attached screenshot where channel monitor screen which looks blurred on top screen. 

 Please help to fix this issue.  Thanks. 

Best Regds
Fayaz

kilrah, 2020-07-30:
No screenshot attached.

fayazmd83, 2020-07-30:
Attached screenshot now pls check

3djc, 2020-07-30:
This is not an x7 access, if you flashed an x7 access firmware then this is wrong

fayazmd83, 2020-07-31:
sorry for typo in description,  my radio controller is Taranis QX7 Accst.  Corrected the description. I have flashed correct firmware by installing opentx companion from here http://downloads.open-tx.org/2.3/release/companion/windows/companion-windows-2.3.9.exe

Followed by flashing process by selecting QX7 from dropdown, please let me know if you need full steps I followed I can send.

kilrah, 2020-08-01:
How is it on other screen pages? It's known that the X7's LCD is poor quality and does this if there are many straight lines in the content.

fayazmd83, 2020-08-01:
I have noticed white columns in few screens due to
which display is affected.  I will have to check all the screens.  Is there
any manual or site where I can test all pages . The Taranis qx7 manual
shows very limited content.   I am a new user so not used to screens
/operations much.

ehasanov74, 2020-08-14:
Hi.
I can confirm same thing.
Channells monitor screen looks like original poster reported.
Other screens looks ok

BRGDS

Edin

fayazmd83, 2020-08-16:
Thanks for confirming.  Any solution identified so far? So , any other
users reported on same issue?

Kind Regds
Fayaz

3djc, 2020-08-16:
Since the other radio run the same code but don’t exhibit this issue, looks like the x7 lcd doesn’t like too much this screen

kilrah, 2020-08-16:
We used to have the borders around the lines removed on that radio because of this, but it seems like that disappeared at some point.

3djc, 2020-08-16:
Brought back.

[image]

closed 2020-08-16 by 3djc

fayazmd83, 2020-08-17:
Dear Developer

Is the issue fixed now?  when is the new release going to be out so that I
can reflash and try to reproduce the problem .

Please suggest.

Thanks
Fayaz

alexdibi, 2020-08-17:
Same on X9 lite[image]

commit 0f9c9a87 (2020-08-16): Do not display too many horintals on x7 screen
This fixes #7854