opentx/opentx #7855: Long Press Menu

mabrungard, 2020-07-31:
With the number of instances when there is a Long Press option for transmitter buttons, it appears that it may be helpful for a sub-menu to pop up that helps identify that a Long Press option exists. Alternatively, maybe the Long Press option is not the best way to move into a selection. Maybe the first press should bring up a sub-menu so that ALL the subsequent options are apparent.

kilrah, 2020-08-01:
Examples?

mabrungard, 2020-08-01:
OK.  

My thought is that instead of a Long Press to get to an alternate menu, it would be better to just employ a short press to get to a root menu where the user can then branch into all the selections of the short press and long press.  

While I'm sure that highly experienced OTX users find the current short and long press menus very quick to navigate, I find that its easy to miss that there is actually a long press menu that I should also be exploring. Branching from a short press menu would assure that ALL users would be persented with all the menus and selections.

jtaylor2, 2020-08-01:
That's not really an example.  I can't really see where a short press on the "enter' button, for example, bringing up a submenu where you have to then select "enter" from the submenu instead of short press being "enter" and long press bringing up a submenu is a improvement.  Same with "menu" on Taranis, short press bringing up a submenu with two choices (system and model) instead of short press bring up one and long press bringing up the other.  Or "page" bringing up a submenu with forward and back instead of short press being forward and long press being back

Perhaps you have a specific example where it would be an improvement, but as a general rule I just see it as taking more keystrokes and resources to do the most common things.