opentx/opentx #7857: GHST (Ghost) Protocol Support

tonycake, 2020-08-03:
ImmersionRC would like to add support for an upcoming R/C link to the OpenTx code base. 

A study was done into the suitability of existing protocols, to avoid the need to do this, but unfortunately, either performance goals cannot be met, or the protocols can not be licensed for 3rd party use.

Protcol name is **GHST** (Ghost), based on raphaelcoeffic's mixer scheduler changes, and hopefully can be released with v2.4.

All code changes have been implemented in a private repository and are ready for review.

kilrah, 2020-08-10:
Hi,
No problem, but please make the repo public so we can all have access to it.

tonycake, 2020-08-10:
No problem at all Andre. So one question... rather new to this open source stuff... the modified files I have are based either on the mixer-scheduler branch, or the tbs-crit-bugfix branch. Since the mixer scheduler stuff has not yet been integrated into 2.4, what is the preferred way to add these files?

3djc, 2020-08-10:
There are two different things:

- for 2.4, someone (Raphael ?) will have to merge crsf shot/multi/ghost handling with the new XJT HD sync, so best for long run is to merge you change in that work. Thats a mandatory step.

- if you need an earlier version, you need to be based on tbs-crit-bugfix, as the initial mixer scheduler branch does have critical  bugs in it, and parner with one of us to have that put in a specific branch. Be aware that that branch is an old derivative of 2.3.5 which does (and will not) include all the 2.3.6 to 9 goodness.

But you need to be aware that:
- you cannot release custom firmware without public sources
- if you want that be called opentx, it cannot be forked and needs to be in our repo

JyeSmith, 2020-08-12:
> 
>     * you cannot release custom firmware without public sources
> 

[image]

Is there a link to the source @tonycake ???

tonycake, 2020-08-12:
you guys are quick :-) Firmware changes are in the same location now. 
For the moment just a zip file containing differences from the tbs-crit-bugfix branch. 
3djc, I would love to get these changes into your repo, but not being a seasoned open-sourcer I would love some more detailed pointers for how to do that. 
Seems like these changes should just be submitted to the tbs-crit-bugfix branch... but seems kinda sacrilegious :-)

3djc, 2020-08-12:
Hello Tony,

Please go to our Discord server (https://discord.gg/CZCwVx2), we can have a more interactive chat there :) (problably will be away from it unitl 10:00 CET)

I did not meant at all that your changes should go in the tbs-crit branch, they  should be on your own branch.

What I meant however is that if you started your work from tbs-crit branch, you should be fine, but if you started from the initial PR (mixer-scheduler), then there are ciritcal unfixed bugs in it and that really should not be used.

Let's try to fix all this today so that you do not find yourself illegal with the OpenTX GPL licence

3djc, 2020-09-07:
This has been merged into 2.3 and imrc firmware source are available in this repository

closed 2020-09-07 by 3djc