opentx/opentx #7858: Value widget in wrong color if no data

loetefix, 2020-08-03:
That the value shows allways in blue if no data is a bug because it should appear in the system "ALARM_COLOR_INDEX".
In default theme it should appear in RED and in the darkblue theme in a grey.

in the file opentx/radio/src/gui/480x272/widgets.cpp there are two lines with:

  lcdSetColor(ALARM_COLOR_INDEX);
i changed to:

 lcdSetColor(lcdColorTable[ALARM_COLOR_INDEX]);
and now the value appears in the right color if no data isreceived

3djc, 2020-08-04:
Not sure it should be alarm colour, need to give it some thoughts

loetefix, 2020-08-04:
Alarm colour is what it was , so i chose it. Alternatively it is possible to make this colour configurable.

PR #7875 (2020-08-15): Fix some colors on value widget
This fixes #7858
closed 2020-08-15 by bsongis