opentx/opentx #7859: 2.3.9 Special Functions options in Companion do not match transmitter

dwheel, 2020-08-03:
**Describe the bug**
When programming Trainer functions in Companion I can not access Parameter options that are available when programming on transmitter.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Special Functions in Companion
2. Click on Any function and select Trainer
3. Cannot scroll to Parameter
4. See Picture

**Expected behavior**
On transmitter, I can scroll over to Functions and select Sticks. Option not available in Companion.

**Screenshots**
See attached picture

**General information**
 - OpenTX version 2.3.9
 - Hardware Radiomaster TX16S and matching Companion
 - OpenTX Companion 2.3.9 Windows 10
[image]

kilrah, 2020-08-04:
[image]

They're just different choices on the main list instead of a separate list.

3djc, 2020-08-11:
Maybe we should just update companion 'Trainer' to 'Trainer Sticks' for better matching with the radio ?

kilrah, 2020-08-11:
Yep that's a good start... and maybe do the same split as on the radio if we feel like.

PR #7876 (2020-08-11): Fix companion "Trainer" to "Trainer Sticks" to match radio
This fixes #7859

There would be a much more work intensive alternative to replicate the two collumns in Compantion, but then that would also affect Set Timers, Override, Adjust GV,....
closed 2020-08-11 by kilrah