opentx/opentx #7869: Add inputs to log replay

3djc, 2020-08-09:
Logs replay allows  to see telemetry input come alive, but while inputs are in the logs, it is currently not possible to see to mixer in action to  troubelshoot model programming errors or things like that.

Log replay should allow to feed inputs

3djc, 2020-08-09:
@elecpower, do you think it is something you could help us with ?

kilrah, 2020-08-09:
I think it's a good thing but there are more important things to improve in the telemetry sim / log player such as loading sensors from the model rather than being hardcoded to the old basic set it currently has...

elecpower, 2020-08-12:
@3djc, I've had a cursory look at the code and it maybe possible but would need re-plumbing with signals and slots to update the radio ui input widgets (sticks, switches, trims) from the telemetry ui which owns the open log file.
@kilrah, agree that should be the ideal but will require significant work.
A common issue/risk is the log file may not match the current radio and model settings. I do not see a solution except to warn the user.
I'm currently working on a PR to implement centralised datamodels to Companion so I wont have time for a while to look at this.

elecpower, 2020-08-12:
I see problems in using for all troubleshooting cases as when replaying the log the simulator may not replicate what actually occurred at the time as the log is a snapshot of all values at the the instant the log entry was generated with no way of determining the actual sequence, delay and duration. For example from one entry to the next SA and SB both changed but did SA change before or after SB, or was the throttle at 0% when SH was held for 3 secs when SF flipped to down (model arming).
I think we would need to manage user expectations about the limitations of replaying a log file.

kilrah, 2020-08-12:
Obviously detail will only be as good as the log frequency that's used. Doubt that's the way to troubleshoot an arming sequence though.

closed 2021-06-03 by stale[bot]