opentx/opentx #7873: Companion Bug when using MASTER/MULTI

jims1234, 2020-08-10:
This issue was discussed recently in www.RCGroups.com's  Multi thread with 3djc and Hpnuts and I was asked to post it here too.  This issue is restricted to MPM based models set up for a new Trainer Mode: called "Master/Multi".
Background: 
I believe issue No 7833 below is directly related. 

To keeping this short I refer you to the DIY Multiprotocol thread Posts 19276, 192771, 19279, 19281, and 19285 for the detailed config and observations. They start with:

https://www.rcgroups.com/forums/showpost.php?p=45169347&postcount=19276
  
At least one other user using this Master/Multi mode is seeing it replicated.  

This issue consistently corrupts data when reading a MPM capable model set up for a Wireless training mode called Master/Multi into the Windows Companion 2.3.9 (and it also impacts the latest nightly version on Companion as of 8/9/2020.)  

Once a corrupted parameter is corrected by manually changing the Model's Radio Page 2 Trainer Mode in Companion read in as: "Master/Bluetooth" back to read "Master/Multi".  The a key problem remains which is:  The Simulator displays in the Trainer OFF state are all just fine.  But, in Trainer ON status, they can't be used because all the TR Channels inputs are transposed and displayed as mis-mapped such that they can't be recovered by transposing the AETR CH 1 to 4 inputs on the Trainer Page of the model again without making the problem worse and messing up the normal 1-to-1 mapping that is Ok in the normal Trainer OFF ops mode..  For example as you move the Rudder stick in the SIM the Elevator Outputs and Ele CHAN Monitor level will move instead.  The Sim displays this mis-aligned mapping ONLY in the Trainer ON mode.  Switching back to the normal non-training mode corrects all CH OUTPUT or CH MONITOR levels and direction displays back to normal. 

 Once the Bluetooth error is corrected in companion, and any other adjustments are made,  anytime a Master/Multi model is written back into the radio it will consistently corrupt the Trainer Mode parameter such that it reads:   "eck RSSI on Shutdo"  instead on the Radio's LCD in place of "Master/Multi" until corrected manually..  it appears occasionally the Failsafe mode set on the Ext MPM can be corrupted as well and it may do this not only one the model  worked but others alongside it as they are saved in the Radio.

The workaround is to always check and correct the Trainer Mode setting in the Instructor's Radio before use with any student radio.  Avoid using the SIM in Trainer ON mode since the displayed channels are mis-mapped. Check and correct the Bluetooth setting to read Master/Multi    
 - OpenTX version 2.3.9 and nightly version 2.3.10N291.exe 
 - Hardware (Radiomaster TX16S using an iRange4+ in Ext Bay under various RF protocols with the model and the INT MPM set into FrSkyRx Rx mode with current firmware in both MPMs)
 - Windows 10 Pro 64bit

3djc, 2020-08-11:
You should not correct the trainer mode in companion, it is a current companion display issue, and if you change the value in companion, this will lead to errors

jims1234, 2020-08-11:
3djc,

I only changed it briefly while troubleshooting but it’s back and set AETR with TR CH 1 to 4 data set sequentially replacing the 100% =: weight in the radios and in Companion.

kilrah, 2020-08-11:
So in a more readable summary:
- Trainer mode modified during import / export (duplicate #7833)
- Trainer channel order not to your liking in simulator - telemetry simu is hardcoded to a particular order which probably won't match your real life setup.

jims1234, 2020-08-16:
A quick update.. I’ve stopped using the simulator’s channel Mixer and Outputs Displays in the sim unless in Trainer OFF state where they are accurate and match the AETR TR ch 1,2,3 4 sequentially I set in both radios, and as long as I change the Trainer mode in the instructor Radio each time from the bogus RSSI string to the Master / Multi Mode it needs, the plane operates just fine in both states: Trainer ON and Trainer OFF So there’s an easy work-around till these two issues are resolved

jims1234, 2021-04-30:
I am sorry this issue went stale since I missed this.. it appears to me that both 3djc and kilrah have mis-interpeted the problem.. The problem still persists.. the problem is not with the Trainer Channel order ( that's a minor aspect of it since I know companion still won't support Simulation in Trainer ON mode..)  The problem is mainly this:  I still need to change the Trainer mode entry manually in the Radio Setup page within the instructor Radio each time I make any change in Companion and Write any model to the Radio.. I then see : "eck RSSI on Shutdo" appear in place of the "Master/ Multi" Mode I had set in Companion and the Radio  needs this changed every time.  The plane then operates just fine in both states: Trainer ON and Trainer OFF.. That has never been solved.. do I need to re-submit it?

closed 2021-07-30 by stale[bot]