opentx/opentx #7879: Companion : ??? in sources instead of EX1/EX2

3djc, 2020-08-14:
2.3.9 and current nightlies

[image]

elecpower, 2020-08-18:
Initial investigations reveal that EX1 and EX2 are set to type None in Radio Settings - Hardware for a new TX16S profile and therefore do not show in the list however when they do appear in the list the two ??? entries still display.

3djc, 2020-08-18:
Just as a reference, it is not TX16S specific, it is the same for x10 for exemple

elecpower, 2020-08-18:
@3djc yes, as I have been working through the code I noticed they share the same logic which is expected. 
Should these radios have JSx and JSy in the sources list? If so then I believe I have the fix.

3djc, 2020-08-18:
No not quiten JSx and JSy are only for X12S where those two slots are used for the "joystick", on other 'horus', it is a spare 'analog) port, like there is SI/Sj for switches

elecpower, 2020-08-18:
If I am understanding you, these two ??? entries should just be dropped from the list.

3djc, 2020-08-18:
When not defined (set to none in radio hardware) then yes they should not be there, howerver if defined, then they should with either custom name or ex1/ex2 name

kilrah, 2020-08-18:
To have the indexes right in other places we normally always show them, but grey out if not applicable

3djc, 2020-08-18:
yes, with ex1 / ex2 name (or custom name)

elecpower, 2020-08-18:
ex1 and ex2 work as they should. Need to prevent 'horus' radios other than 12S displaying ??? in the joystick reserved positions

3djc, 2020-08-19:
Just display ex1 ex2 instead of ??? (since ex1 for x10 is on the same slot as JSx for x12). In other words, always show for horus but ex1/2 when !x12s, JSx/y if x12s

PR #7888 (2020-08-28): Companion fix sources lists for X10 and T16 radios and improve radio conversion of sources
Fixes #7879
Improve sources conversions between radios eg #7624
closed 2020-08-28 by bsongis