opentx/opentx #7882: Change or adjust OpenTX Font color

4dimensio, 2020-08-15:
Good morning,
 I have a Horus X10S and I am running OpenTX 2.3.x and using the "Default" theme. I have changed the background picture and it is dark and it is very hard to see the fonts,words,Trims hash marks or any data on any tele screen...ect.
Is their anyway I can make these fonts,words,Trims hash marks "white" like in the Darkblue" theme so I can see them?
Adjusting the "Background" color doesn't help....font and such still is black
Adjusting the "Main" color doesn't help either......font is still black
Thank you ,
Brian

3djc, 2020-08-15:
We have released recently a new flexi theme available the nightly sd pack.

closed 2020-08-15 by 3djc

4dimensio, 2020-08-15:
Great, I will check it out and hopefully it allows me to change fonts  color