opentx/opentx #7893: External Radio Module Receiver No Field

miguel-orange, 2020-08-20:
In the upcoming future, external radio module protocol at least for Crossfire will have a Receiver No. I'd like to request that field be available/mapped when editing a model.
It might makes sense to also make sure the Model summary page, column RX # also displays Receiver No. for external modules like it does now for internal radio system.

[image]
[image]
[image]

3djc, 2020-08-20:
The first part is within the crsf modelid PR.

On the last screen however, it poses a lot of issues since both internal and external can have their own number, so which one do you display, it raises a lot of issues

miguel-orange, 2020-08-20:
label and delimit? Acceptable values could be `01-internal` or `02-external;01-internal`.

elecpower, 2020-08-24:
What about separate columns "Int.RX #" and "Ext.RX #"? This would also make identifying and colour coding duplicates simpler.

3djc, 2020-08-24:
sure

closed 2021-06-18 by stale[bot]