opentx/opentx #7899: 2.3.9 - PPM FASST module it does not initialize in 7CH mode only on X10S Express

robustini, 2020-08-25:
I would like to understand if it is a hardware limitation of the X10S Express, since with the Radiomaster TX16S and Horus X12S "Express" (with ISRM) it works perfectly with the same OpenTX firmware version.
The FASST module is correctly set in the OpenTX settings, but as you can see from the first video it is not initialized in 7CH mode (alternating red and green led), while in 8CH mode the led quickly blinks green (second video, waiting for bind).
How does the X10S Express differ in not making it work?
OpenTX issue oer hardware limit?

https://youtu.be/T0jxEWdagMk

https://youtu.be/MPV4A4VBqGM

**General information**
 - OpenTX version V2.3.9
 - Hardware Horus X10S Express - Horus X12S Express - Radiomaster TX16S

kilrah, 2020-08-25:
Do you have an oscilloscope to compare the signals on both (with module connected)? Won't really be possible to know without seeing both.

robustini, 2020-08-25:
> 
> 
> Do you have an oscilloscope to compare the signals on both (with module connected)? Won't really be possible to know without seeing both.
> I remember with that module it was required to play with polarity and pulse length until it was happy.

Hi @kilrah, thanks for pointing me in the right direction.
Unfortunately I don't have an oscilloscope, but surely the X10S Express sends out something in PPM different from the other TXs.
I tried playing with polarity and pulse length with no success.

robustini, 2020-08-25:
Found it!
Just put 350us instead of 300!

closed 2020-08-25 by robustini