opentx/opentx #7901: Linux USB HID joystick emulation

minzukas, 2020-08-25:
Radio: Radiomaster TX16s with OpentTx 2.3.9
Ubuntu 20.04.01 LTS

When usb connected radio prompts screen HID/storage. Whichever selected I get 2 mass storage devices in linux unable to detect joystick and map it in simulators.

3djc, 2020-08-26:
Cannot reproduce

jtaylor2, 2020-08-26:
Can not duplicate on openSUSE 15.0 either.  Are you sure you tried both?  The darker one is the sellection.

`Aug 26 07:39:20 HP-Linux kernel: usb 8-3: Product: RadioMas TX16S Joystick
Aug 26 07:39:20 HP-Linux kernel: usb 8-3: Manufacturer: FrSky
Aug 26 07:39:20 HP-Linux kernel: usb 8-3: SerialNumber: 00000000001B
Aug 26 07:39:20 HP-Linux kernel: input: FrSky RadioMas TX16S Joystick as /devices/pci0000:00/0000:00:12.0/usb8/8-3/8-3:1.0/0003:0483:5710.0003/input/input16
Aug 26 07:39:20 HP-Linux kernel: hid-generic 0003:0483:5710.0003: input,hidraw2: USB HID v1.11 Gamepad [FrSky RadioMas TX16S Joystick] on usb-0000:00:12.0-3/input0`

minzukas, 2020-08-26:
I found a solution that works, though it is still strange. I have to go to settings on the radio and choose joystick instead of ask. When connected to pc nothing happens. Then I change setting to ask while connected and radio becomes discovered as joystick.
Workaround :)

kilrah, 2020-08-26:
If you set it as joystick in the menu there's no point switching to ask again, it'll jsut always be joystick.

And yes, it seems you just selected the wrong line when it was on Ask like many people do.

closed 2020-08-26 by kilrah

jonmahoney15, 2021-03-06:
I am having a the same issue, I know I am selecting the correct choice in the ask menu, I am also able to see my quadcopters connect over the usb-c connection.  I am on Ubuntu 20.04 LTS using radiomaster as well. I am unsure if I am just missing some library on my Ubuntu though. No matter if I use joystick or ask, controller does not show up in my jstest-gtk

ligname, 2022-01-15:
Hi everyone, i am facing the same issue from above: i am using a radiomater tx16s on my pc linux with devian 11 but i can't get any responce attaching the usc cable, dmesg is always giving : 
1383.733036] usb usb7-port1: attempt power cycle
[ 1384.709309] usb usb7-port1: Cannot enable. Maybe the USB cable is bad?
[ 1385.609325] usb usb7-port1: Cannot enable. Maybe the USB cable is bad?
[ 1385.609378] usb usb7-port1: unable to enumerate USB device

the same cable/radio is working on my mac.
there is any patch or library to install?

regards