opentx/opentx #7917: Radiomaster TX16S and Frsky R9M, ACCESS registration not possible

dl1bg, 2020-09-03:
Now successfully running my TX16S with Frsky R9M 2019 and R9MM-OTA on ACCST. Yes, downgrading is no problem, just trust Pawel S :-)

But with the latest receiver, the R9MX, we have to go to ACCESS and perform the 2 stage binding process:
https://www.frsky-rc.com/wp-content/uploads/Downloads/Manual/R9%20MX/R9%20MX-Manual.pdf

Even with the latest nightly build, there is no possibility to perform the registration of the receiver. "Bind" only does not work.
Could you please implement this for the TX16S? For other TXs it seems to be available.

Pls tell me if I can help testing, hardware and measurement equipment is available.

Btw, RX firmware update via OTA is also not availlable, but S-Port update can be used as workaround.

3djc, 2020-09-03:
Only FrSky radio that natively do ACCESS can do R9M ACCESS. So TX16S won't do access either

closed 2020-09-03 by 3djc

Pick86, 2021-12-17:
I'm running access on the tx16s via internal access mod with the 2019 modula latest open tx no problems with ota and my r9mx recievers work great only problem so far is fighting with telm in inav I think the answer is right in front of me still pretty noob but picking it up slowly.