opentx/opentx #7922: ACCESS Registration ID editable in Companion request.

Hobby4life, 2020-09-06:
It would be great if the ACCESS registration ID can be editable in Companion

3djc, 2020-09-06:
Not sure how, since it is transferred in the receiver at registration time.

Hobby4life, 2020-09-06:
The issue now is.. If I copy my model over to another ACCESS radio, and want to register the RX to that new radio. The old ID is brought up to register, not the ID of the new radio.

3djc, 2020-09-06:
If they are both your radios, why don’t you set them to the same Id ?

Hobby4life, 2020-09-06:
Correct that is what I have done now, but then again If the old ID is abcd1234 and the new ID is hobby_4l, registering keeps popping up with abcd1234 from the old radio because that was stored in the model.
Registering should always read the TX ID that is set on that specific radio.. now i does not.. it used the last ID used in the model.

3djc, 2020-09-06:
It does, UNLESS a different one from the default has been set and sent to the receiver. Editing it afterwards would bring tons of confusion imho

Hobby4life, 2020-09-06:
I Agree editting afterwards would confuse a lot.. forget about that, but still when I copy a model from radio ABCD1234 to radio HOBBY_4L and then register again on TX with HOBBY_4L, it wants to register to ABCD1234 because that was used in the model..
Entering registration does not read th TX ID from the TX itself, it reads the ID from the model.. thats not how it should work.

Alex00101, 2020-09-07:
May I support this improvement/fix as it was my experience as well.

closed 2021-06-18 by stale[bot]