opentx/opentx #7923: Better RSSI spike filter.

Hobby4life, 2020-09-07:
I have always noticed due the effect of multipathing RSSI spikes over time , that is not a problem. that is very normal.
(X9D plus, X10S,X10S Express, Xlite, Xlite-pro, X12S, X12S access)
But triggering the low signal alert everytime an extremely sharp spike is detected is very annoying.
I can set the low signal warning for access lower then 35 and the problem is somewhat put back in the background.
A better solution would be to only trigger if the spike is longer then X uS/mS

Maybe adjustable in the telemetry screen?

3djc, 2020-09-07:
You can create such an alarm if you desire using LS and SF, but forget your uS/mS units. Rssi comes from receiver using the telemetry feed, and values do not change that quickly

Hobby4life, 2020-09-07:
Hmm you are correct, RSSI ofcourse is sent out by the RX, so no change is possible on the TX in that matter then.

BTW since VFR is introduced, I only use that as a readl valid signal indication.
When these spikes occur, mostly the VFR  values are +90%, even when RSSI is low <50 then are usually +90%

What I can do is put RSSI alarms down to 0, and create my own then.

closed 2020-09-07 by Hobby4life