opentx/opentx #7925: Tele2 Recvr Bat Voltage Ratio Fluctuates

CaptLeMay, 2020-09-08:
Radiomaster TX16S setup per the Users Manual.
Firmware:  opentx-tx16s-bluetooth-faichoice-flexr9m-lua-ppmus-                   en-2.3.9-otx.bin
                 multi-stm-opentx-taer-noinv-v1.3.1.49.bin
FrySky D8R XP Receiver
3S GensAce Lipo Flight battery

Tele A2 Flt Bat Voltage Ratio varies each time the receiver is powered up.  It usually starts around 24 to display proper voltage, but drifts down to about 8.6 as time goes on.

[image]
[image]

3djc, 2020-09-08:
Telemetry is dispalyed untouched, so it is an issue on your receiver side imho

kilrah, 2020-09-08:
Did ysou actually connect a sensor to the A2 input? The receiver doesn't magically guess the pack voltage.

closed 2020-09-08 by kilrah

CaptLeMay, 2020-09-12:
Andre, Thanks for your reply.Attached is the instruction sheet for the receiver.As you can see on the instruction sheet, there is an internal sensor for the battery voltage with a 4:1 ratio. 
 Wow!  It must be Magic!!  I call it firmware.I did not attach an external sensor since the receiver has an internal sensor.
I will get an external sensor and see if that works without fluctuations.If it still fluctuates, would you say this may a software issue?  Admittedly, it may be a problem with the receiver. I have 2 D8R II receivers and the fluctuations occur on both.
Thanks for your time.Ross

kilrah, 2020-09-12:
The internal sensor gives you the voltage the receiver is supplied with , either as RxBt or A1.

A2 is an external input for a sensor, and if you want to measure pack voltage you need to use that. The receiver doesn't have access to the pack voltage without one.

CaptLeMay, 2020-09-12:
OK,I'll put an external sensor on.