opentx/opentx #7927: "Slow" functions slows input channel, not the Mix.  ei: aileron slow, not flap

CaptLeMay, 2020-09-08:
Radiomaster TX16S setup per the Users Manual.
Firmware:  opentx-tx16s-bluetooth-faichoice-flexr9m-lua-ppmus-                   en-2.3.9-otx.bin
                 multi-stm-opentx-taer-noinv-v1.3.1.49.bin
FrySky D8R XP Receiver

Mixing flaps with Ailerons, the "Slow Up/Down" function does not slow the flaps, but it does slow the aileron movement.  Also, must use the "Replace" function to prevent an opposite direction movement of the ailerons prior to the correct direction movement.  IE:  Moving the aileron up first causes a down movement followed by the proper direction up movement.

[image]
[image]
[image]
[image]
[image]
[image]

3djc, 2020-09-08:
This place is not for end user support. Your programming is wrong in many ways, but this is not the right place for discussing it. Please go to community sites like rcgrousp or openrc forums for community support

closed 2020-09-08 by 3djc

CaptLeMay, 2020-09-08:
Thank you for your reply.
Sorry to have posted on the wrong site for assistance.
As I obviously need help in the programming,would you please direct me to a specific site or sites where I can get it. 
 I have read through most of the openTX university documents, and watched multiple YouTube videos to try to learn.  I have found nothing relating to the problems I am experiencing with the slow; dif, and travel adjustments.