opentx/opentx #7933: GPS in logfile

WimSanders, 2020-09-11:
I Have a Horus 12 S.  
Is it posible to add the GPS data from the transmitter to the logfile?

lshems, 2020-09-17:
You could do that using a lua script, but I agree it would be logical to consider a sensor as a sensor, on the radio or on the plane.

If you really need it quick, PM me on Rcgroups

WimSanders, 2020-09-18:
I agree with you a sensor is a sensor undipening if it’s a transmitter or receiving sensor
I’m building  my own log analyzer who display also other  thinks the GPS position off the plane
Depending on the time of the day (sunset) the transmitter position change on ore field.     
The GPS data from the transmitter is a “nice to have” option. 
Not so many open TX have a Horus X12 transmitter.
Maybe in a next, next…. release its possible  to implement this option
The trim setting is match more impotent. 
How can a analyze my steering commands if I don’t know the trim settings
I think this is not only important for the X12 users but for all users.
Off topic of this request:
I would like to learn lua, but don’t  know where to start.
             Where to start
             Do I need a compiler?
           What are the typical  open XT commands
With kind regards,
Wim Sanders

Hobby4life, 2020-09-24:
I agree too, make the intGPS available as sensor

closed 2021-06-26 by stale[bot]