opentx/opentx #7934: No Frsky X option on 2.3.9 TX16S

Daviot52, 2020-09-12:
2.3.9 Open TX 
Radiomaster TX16S
No option for Frsky X only X2

kilrah, 2020-09-12:
Might need to update module.

closed 2020-09-12 by kilrah

Daviot52, 2020-09-12:
Module just updated yestarday. Reflased rx to fcc v1 tried option Frsky D16 and it bound ok. Issue seems to be change in OpenTX as option for X8R is now Frsky D16.  Option Frsky X no longer available.

3djc, 2020-09-12:
FrSky X never was in OpenTX, we rename it

Daviot52, 2020-09-12:
So difference in multiprotocol module naming and open tx naming ?

3djc, 2020-09-12:
X, D and V are regrouped as one for all of thos, and you select them as subtypes. X2 is not included tho

Daviot52, 2020-09-13:
Thanks for the explanation. I was more than a little confused due to the
similarities with the multimodule but not seeing the same structure in open
tx. Is this listed somewhere ? I had done several searches but not come
accross this information. I also had some older X8Rs that did not seem to
bind with any protocol which helped increase the frustation reflashed them
now all is ok.
Thanks.

kilrah, 2020-09-13:
It's in the MPM doc.
https://www.multi-module.org/using-the-module/protocol-details/frsky-d