opentx/opentx #7935: Frsky R9m access for Rediomaster TX16S & jumper T16

cvetaevvitaliy, 2020-09-13:
Hi, I made this mod a long time ago and use it, it aroused a lot of interest among people, I will get a lot of reviews and emails about it (not all people write on GH, because they do not know how to use it)
can I do a PR and merge a branch with the OpenTX repository?
So that they have the opportunity to download the firmware from the companion.
will this be interesting for OpenTX?
after all free software and GPL license

P.S. I can make a normal instruction for WiKi, since I did everything in haste

https://github.com/cvetaevvitaliy/Radiomaster_TX16S
Source code
https://github.com/cvetaevvitaliy/opentx/tree/RadiomasterTX16S_ACCESS

al21883, 2020-09-19:
I support that idea. It will be nice to have "official" firmware with R9M support.
I'm learning too how GitHub works. I thought that you was part of tje OpenTX team, and merging will be done on the next release.
I assume you already thought about it, but... It would be any way to get ACCESS support without soldering those 3 wires? Maybe with major firmware modifications?

Thanks for your nice contribution!

3djc, 2020-09-19:
It cannot be done without hardware modification because FrSky made the R9M 2019 use different pins in ACCESS mode

cvetaevvitaliy, 2020-09-19:
I can try to do without hardware modification, for R9M to work, needs hardware uart.
I can try implement soft uart, need two dma channels and a free timer

3djc, 2020-09-19:
No, we will not allow softserial at 450k

cvetaevvitaliy, 2020-09-19:
in any case, if the user does not want to disassemble the radio, he can connect his R9M to the Uart that is brought out from the bottom of the equipment case, they are available there.

anyway, it is open source software,
mod does not use FrSky ACCESS library, what violates the GPL, I shared my work and upload software.

Thanks to the opentx team for OpenTX

cvetaevvitaliy, 2020-09-19:
> I support that idea. It will be nice to have "official" firmware with R9M support.
>
> I'm learning too how GitHub works. I thought that you was part of tje OpenTX team, and merging will be done on the [...]

No, I'm not part of the OpenTX team.  
I can try to do it without hardware modification, the main thing is to do it well.
STM32 has doc on how to implement SoftUART

needs to be tested higher speed than 115200

https://www.st.com/en/embedded-software/stsw-stm32156.html

al21883, 2020-09-20:
> No, I'm not part of the OpenTX team.
> I can try to do it without hardware modification, the main thing is to do it well.
> STM32 has doc on how to implement SoftUART
>
> needs to be tested higher speed [...]

Maybe this speed on softserial is too much load for the processor and don't worth the cost for this feature. 
HW modification may not a big deal for most of advanced users who used to play with RF modules.
If you've tested it, I vote for merging it as is with master for the next relase.

3djc, 2020-09-20:
We are going to have a look at our own implementation of this (using the HW mod, not softserial)

al21883, 2020-09-20:
> We are going to have a look at our own implementation of this (using the HW mod, not softserial)

Thanks a lot!

cvetaevvitaliy, 2020-09-20:
> We are going to have a look at our own implementation of this (using the HW mod, not softserial)

Please look and test.
there are not many changes

https://github.com/opentx/opentx/compare/2.3...cvetaevvitaliy:RadiomasterTX16S_ACCESS

cvetaevvitaliy, 2020-09-20:
can the Radiomaster team make this modification in the next PCB versions,
they only need three lanes on the board, no new components are required.

3djc, 2020-09-20:
Since this mod does prevent/break usage of one of the bottom UART connector, it cannot be pushed onto everybody. It is a choice users have to make. Many have absolutely no use for R9M ACCESS because they do not do long distance, or much prefer Crossfire or Ghost

cvetaevvitaliy, 2020-09-20:
> Since this mod does prevent/break usage of one of the bottom UART connector, it cannot be pushed onto everybody. It is a choice users have to make. Many have absolutely no use for R9M ACCESS because they do not do long distance, or much prefer Crossfire or Ghost

The user should always have a choice, not everyone can buy TBS Crossfire or ghost.  Although I have both TBS and R9M and my Ghost comes to me with Bangood

It is not for us to decide which of the modules is better: TBS or R9M, the user will decide. 

User can download this firmware and flash, just like: flash the firmware from a companion with Lua or without Lua, for example

He can download this firmware, just like: flash the firmware from a companion with Lua or without Lua

I'm not a supporter of either R9 or TBS or ghost, I'm pilot, embedded developer and I liked fly 😊

Since it is free software - I uploaded it to others people.

The user should always have a choice, not to throw away his R9M to trash 😟

3djc, 2020-09-20:
Exactly, so why should Radiomaster forçe that in future versions ???

cvetaevvitaliy, 2020-09-20:
I just suggested that, to implement it, it doesn't cost anything, no new components are needed.

As for me, it is attractive from the side of the business and from the client side.  excellent radio and supports all possible external XJT modules

3djc, 2020-09-20:
But it prevents usage of one of the bottom uart, so you would be choosing for users ...

cvetaevvitaliy, 2020-09-20:
this is a mod, so the advertising company was positioned: there are two uart do what you want :)

cvetaevvitaliy, 2020-09-20:
Radiomaster, for example, can make a small PCB adapter (place two inverters) and also sell them in his store 
these are trifles and care about users

suddenly FrSKy will take its head, find new developers and release an update that will be better than TBS 😆
hardware is a good module on Lora SX1278

TedAus, 2020-09-21:
Please include this as an option (like Lua, Bluetooth, R9Flex, etc.) in the next release. I'm using it and am very appreciative that I can finally utilise all of R9M's features. However, by using this, I have lost the Bluetooth option that is available on the stable version. Also, I'm getting sensor lost with Crossover/Oversky (Frsky D16) receiver. So it would be great to have R9M together with Bluetooth and the reliability of the stable release. 

Another nice to have is the Auto power level for R9 Flex. It's not clear if cvetaevvitaliy's version supports auto power switching up to 1W.

Thanks all for your work. Very much appreciated.

brainbubblersbest, 2020-09-25:
Hello, i ask about the same possibility for Non Express Horus Transmitters. #7610

3djc, 2020-09-26:
I cannot really understand your long post when this (https://github.com/opentx/opentx/pull/7963) is in progress....

cvetaevvitaliy, 2020-09-26:
Good, thanks 👍🏻

cvetaevvitaliy, 2020-09-26:
I think I need to update WiKi and give instructions on how to connect and use it

3djc, 2020-09-26:
There is a wiki here, are you able to edit it ? https://github.com/opentx/opentx/wiki/Externalaccessmod

cvetaevvitaliy, 2020-09-26:
Nop

brainbubblersbest, 2020-09-26:
> I cannot really understand your long post when this (#7963) is in progress....

Because i request this for Non Express Horus Transmitters, and if you take a look at #7610, than you can see:

"For This it can be used aux_serial (1)
Aux2_serial is still used for Bluetooth serial(USART6)

The x10 express Pins PB10 and PB11 from Stm32 are used for external access (USART3) . In x12s AND x10s cases, this Pins are used for the p13 connector aux serial. From here a trace to jrbay is easy to realise.
This connection is disabled outside a debug OTX Build anyway."

Sorry, i just want to argument and explain on these request. And i dont realized that this is already in Progress.
But i can make a PCB Pic shot with the extra Wiring from Horus, for the Wiki and other Users out there.

kilrah, 2020-09-26:
This is for any radio that has a spare uart.

cvetaevvitaliy, 2020-09-26:
> > I cannot really understand your long post when this (#7963) is in progress....
> 
> 
> 
> Because i request this for Non Express Horus Transmitters, and if you take a look at #7610, than you can see:
> 
> 
> 
> "For This it can be used aux_serial (1)
> 
> Aux2_serial is still used for Bluetooth serial(USART6)
> 
> 
> 
> The x10 express Pins PB10 and PB11 from Stm32 are used for external access (USART3) . In x12s AND x10s cases, this Pins are used for the p13 connector aux serial. From here a trace to jrbay is easy to realise.
> 
> This connection is disabled outside a debug OTX Build anyway."
> 
> 
> 
> _I thought that this Information is helpfull for u, to code this function on Aux1_Serial (USART3)
> 
> beside the Factory installed BT Module on the Aux2_serial (USART6)_
> 
> 
> 
> Sorry, i just want to argument and explain on these request. And i dont realized, that this is already in Progress.
> 
> But i can make a PCB Pic shot with the extra Wiring from Horus, for the Wiki and other Users out there.
> 
> I really want to help, if i can.
> 
> 
> 
> Just to clearify,
> 
> Since Kilrah commented my old #7610 as a duplicate to this, 
> 
> does that mean this request is provided for the X10 and x12 Radios too?
> 
> Can provide with testing and Documentation on x12s only.

I would help, but I do not have this equipment for testing, it is expensive and I see no reason to buy

you need to find these pins PB10, PB11
Maybe this pin go to the P2 connector,  see the bottom of the board, where the SD card is

[image]

brainbubblersbest, 2020-09-26:
> This is for any radio that has a spare uart.

OK, to be more specific, since UART1 (USART3) is been disabled since #6733
Does that mean this is not a free Spare UART anymore, and probately never was?
Can this be Changed in Handling familiar to X10s Express TX by HAL modification?
Thats my concerns about these.

cvetaevvitaliy, 2020-09-26:
[firmware_x10_access_uart3.zip](https://github.com/opentx/opentx/files/5286832/firmware_x10_access_uart3.zip)

brainbubblersbest, 2020-09-26:
> [firmware_x10_access_uart3.zip](https://github.com/opentx/opentx/files/5286832/firmware_x10_access_uart3.zip)

I already measured that these Pins from the Processor goeas straight to the P13 Connector. (AUX1)
(as i wrote above)
My concerns about these implemantation where related to #6733
And if the Factory BT connected to AUX2 needs to removed on Horus or can be leaved in place.
Because different then on Horus.

AUX2 is USED for that on the RM and Jumper Code u created.

cvetaevvitaliy, 2020-09-26:
This P2 ? ![Uploading 94343644...]

brainbubblersbest, 2020-09-26:
[image]
This is Serial_1 on x12s

cvetaevvitaliy, 2020-09-26:
Sorry, I thought you had X10 You just write so much that it's hard to understand you)

brainbubblersbest, 2020-09-26:
> Sorry, I thought you had X10 You just write so much that it's hard to understand you)

Sorry for that, I just want not to, forget providing any Information that could be helpfull.

cvetaevvitaliy, 2020-09-26:
Find where these PINs go
[image]

cvetaevvitaliy, 2020-09-26:
> > Sorry, I thought you had X10 You just write so much that it's hard to understand you)
> 
> Sorry for that, I just want not to, forget providing any Information that could be helpfull.
> Ur provided Firmware can be used for X12?
> I read in the Hal, that there are some Differences between x10 and x12?
> In the meantime i make the modification ;D

Download this firmware and look in the menu EXT module, there will be a new item  - R9M ACCESS

[firmware_X12S_ACCESS_EXT.zip](https://github.com/opentx/opentx/files/5286967/firmware_X12S_ACCESS_EXT.zip)

brainbubblersbest, 2020-09-26:
Okay. I measure this already. As documented in #7610
But I double Check this for u with a prove of measurement.
I know this have to be sure that it is correct. 😉 
And I checked also that there is NO short to vcc or gnd.

UART_3TX goes to P13 TX
[image]

UART3_RX goes to P13_RX
[image]

NO resistors or else between this measure points.

cvetaevvitaliy, 2020-09-26:
You need 2x inverter for RX and TX
Include them in the gap between RX and TX wires and solder to module

[image]

cvetaevvitaliy, 2020-09-26:
If you show a photo of the board where the EXT module is connected, maybe I will find it

cvetaevvitaliy, 2020-09-26:
I checked the firmware that was previously defective, there is no new item in the R9M ACCESS menu! So I built a new FW and checked it in the simulator, everything is OK!

[image]

[image]

[firmware.zip](https://github.com/opentx/opentx/files/5287071/firmware.zip)

brainbubblersbest, 2020-09-27:
I measured and documented some stuff for the Wiki, this can be simplyfied later for add it to the Wiki .
Because of that, i think the Heartbeat Protection from #7963 is not in need. But maybe, i oversee something here.
I measure the Traces on the Mainboard itself Tomorrow.
For TX we can use the existing Inverter that generates the SBUS inversion.
For RX we need for sure a external one in addition.
[image]

brainbubblersbest, 2020-09-27:
I located the Inverters, but im noticed that the original traces have resistors between the f4 and the inverters.
You go straight to the Gateon your mods for RM and Jumper, is that right?
For the External Inverter its easier for other Users to use a available Board to solder.
I Attached two of them and already have these two, do you think we can use one of these two?
[image]
[image]
[image]
[image]

cvetaevvitaliy, 2020-09-28:
> I located the Inverters, but ive noticed, that the original traces have resistors between the f4 and the inverters.
> You go straight to the Gate on your mods for RM and Jumper, is that right?
> For the External Inverter its easier for other Users, to use a available Board to solder.
> I Attached two of them and already have these two, do you think we can use one of these ?

I see that the circuit is not very different from Radiomaster and Jumper, you can do this, these are inverters
The blue wire (third) must be soldered, because the diode at such high speeds UART does not have time to open

[image]

brainbubblersbest, 2020-09-29:
Hello, the TX is Good. But the RX soldered this way affects the Horus Heartbeat, if FRSKY Internal is Activated In Dual Mode with the External Access. So the external Mod is definately needed. So it doesnt leads to more Problems.
(thats why the New express X10 has a pin More to connect the HF Mainboard)
I finally found a little Breakoutboard for 1-2€.
Easy to solder and available. 
This isnt a Problem with internal 4in1, but on Horus Radios.

cvetaevvitaliy, 2020-09-29:
> Hello, the TX is Good. But the RX soldered this way affects the Horus Heartbeat, if FRSKY Internal is Activated In Dual Mode with the External Access. So the external Mod is definately needed. So it [...]
It does not affect Heartbeat.
You can do as I showed in the picture, or you can not for activated R9M access

brainbubblersbest, 2020-09-29:
How can you be sure? You will use the inverter that Trace the "clock" Signal from ixjt, used for timing for a better latency? I will try it as you described. 😉
If this is not working in a redundancy SETUP, with 2,4 and R9M ACCESS enabled the same time. Then i can still Do it with a additional inverter.

Btw, i have also Crossfire and a Isrm Upgrade Module.
So i need these options: Crossfire, lua, luac, noheli, multi. And the Code from this😉
I can test this the next two Days.

cvetaevvitaliy, 2020-09-30:
does r9m access work if you powerOn only external module?

TedAus, 2020-09-30:
Yes it does on mine. I always disable the internal module when using R9M.

cvetaevvitaliy, 2020-09-30:
why power on internal when to use external?

brainbubblersbest, 2020-10-01:
Im done with the wiring!
You can use 2,4 and r9 receivers together as an "Dual Band Receiver Redundancy" Setup. 
The Receivers are linked to each other over its s.port pins.
I used the external Inverter that is used for the QX7 Inverter Mod.

I noticed my X12 need the QX7 Crossfire Mod also, because off the repeately Telemetry loss warnings and Lua Errors.
The Inverters that are not Capable of doing 450.000baud have all the J6 on it.
The Inverters in X10S too. [Inverter Mod for 400.000baud](https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7/#Mod_fuer_FrSKY_Horus_X12)

The Jumper and Radiomaster Models u modded have all the faster Y3J  or J1 Inverters.

Because R9mACCESS Use the Serial Connection also at 450.000 baud and these low Speed Inverters are everywhere in X10 and X12S. I decide to use 2x Qx7 Crossfire Mod Inverters for the Serial Inversion and solder Cables direct to the HF Board.
This was easy to do and now i have separate Wiring.
No Connection to internal Heartbeat or the old CPPM trace.

Same Wiring should work on a x10.
Can you make a Build with: 
-Crossfire, 
-Lua, 
-4in1, 
-noheli, 
-Internalaccess and 
-Externalaccess? 
So i can test that everything work as it should.
I will test every Configuration, and Tomorrow i post a wiring plan for this.

brainbubblersbest, 2020-10-01:
Okay. I flashed your last posted firmware and it wont boot. 
Display is dark, and no Module in the Bay.
Back to 2.3.9 and it boots again. Try more tomorrow.

zzsdeo, 2020-10-01:
@cvetaevvitaliy 
I made a modifications described in your repo and flashed a firmware with access support but on the info screen under Modules / RX version I see "---" instead of actual module version. Receiver is not recieved from AliExpress yet and I cannot bind it. Is it happens because I didn't bind the RX or I mistook in wiring or something else?

cvetaevvitaliy, 2020-10-02:
first you need to bind the receiver and it will appear there

cvetaevvitaliy, 2020-10-02:
> Okay. I flashed your last posted firmware and it wont boot. 
> 
> Display is dark, and no Module in the Bay.
> 
> Back to 2.3.9 and it boots again. Try more tomorrow.
> 
> Maybe this is related to that code?
> 
> https://github.com/opentx/opentx/commit/a083342cda7ec0ecbece7fca1aeabd5377216b1b

I can only do this on weekends, since I work on weekdays

Pigost, 2020-10-03:
> This P2 go to the AUX1 (UART3 PB11, PB10) ?
> 
> [image]

Yep! I checked them out! Horus X10S.
[stm32f439bi.pdf](https://github.com/opentx/opentx/files/5321419/stm32f439bi.pdf)

al21883, 2020-10-03:
Tested on my TX16S And working perfect. Thank you very much @cvetaevvitaliy !

I don't understand why this issue isn't a Pull request.

3djc, 2020-10-03:
There is one ....

Pigost, 2020-10-03:
Horus X10S (non ACCESS)
[image]

Pigost, 2020-10-04:
The mod
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

 with FrSky Horus X10S is complete!
Everything works fine! Many thanks!

[115 more replies]

closed 2020-12-20 by 3djc