opentx/opentx #7936: Contact Debouncing power button

rcmansid, 2020-09-13:
There is a Power off issue with FRSky Taranis Qx7 Access transmitters that need to be address. I have a New Qx7  Bootloader Version 2.30 it worked properly. When I installed my model settings from an older Qx7 and put in a new IRx4 multi module it said I needed to update the new module and would not let me switch to DSM2 . I did not want to do that . I  installed OpenTX  Version 2.2.4 on the new transmitter. It worked until I tried to turn it off . The  power button keeps cycling from on to off at about 0.5 hz . The only way to stop it is forcing it into Boot mode or pulling the battery. How can I fix that myself ?

jtaylor2, 2020-09-13:
The X7 Access was not supported on 2.2.4.  The first release to support the X7 Access was OpenTX 2.3.3 https://www.open-tx.org/2019/12/20/opentx-2.3.3  The only way to fix it is to upgrade your radio to a supported version (2.3.9).

rcmansid, 2020-09-13:
Thanks so much good to talk to an expert. If I mount 2.2.9 will putting the Ver 2.2.9.bin into the transmitters EEprom and not the SD card? If I do it from Companion it will not let me do the upgrade now that 2.2.4 is installed.
 Thanks so muchfor the wisdom..
Sid Harden

rcmansid, 2020-09-13:
Thanks so much good to talk to an expert. If I mount 2.3.9 will putting the Ver 2.3.9.bin into the transmitters EEprom and not the SD card? If I do it from Companion it will not let me do the upgrade now that 2.2.4 is installed.
 Thanks so muchfor the wisdom..
Sid Harden

    On Sunday, September 13, 2020, 05:26:15 PM CDT, Sid Rcmand <rcmansid@yahoo.com> wrote:  
 
  Thanks so much good to talk to an expert. If I mount 2.2.9 will putting the Ver 2.2.9.bin into the transmitters EEprom and not the SD card? If I do it from Companion it will not let me do the upgrade now that 2.2.4 is installed.
 Thanks so muchfor the wisdom..
Sid Harden

3djc, 2020-09-14:
Closing as this is obviously not an OpenTX bug

closed 2020-09-14 by 3djc

rcmansid, 2020-09-18:
Not closed as far as I am concerned. The Qx7 Asst Transmitter turns off properly when you pull the trims to convert it to Bootloader mode. It is still running the firmware at a different location in the firmware program. Once in the bootloader mode power down is instiant. There is an error in the delayed countdown feature in control mode of the firmware. If you want I will make a video to prove my point. I have 30 years of imbedded hardware/software design and I am pretty certain that the code can correct the problem that FRSKY has created in their hardware. I am not the only one that has had that issue.
I really like OpenTx and thrilled how it is advancing the state of the art in Radio Control. To bad the other makers in the USA and elsewhere have not joined in. Keep up the fantastic work.
Thanks for your response and patience.

Sidney HardenSenior Software Engineer, Basler Electric , Highland , IL (Retired)

kilrah, 2020-09-18:
Yes, it is "fixed" by using the right firmware, for the ACCESS version, which means at least 2.3.3.

rcmansid, 2020-09-20:
Thanks for the answer about reinstalling Ver 2.3 or above. Now with 2.2.4 running in the Qx7 Access I get a version error if I try to install Ver 2.3.9. Can it be done via the SD card root ?   Thanks again:
Sid Harden

kilrah, 2020-09-20:
Yes, or untick "Check hardware compatibility".