opentx/opentx #7947: OpenTx Companion downloads wrong version of firmware

querist, 2020-09-18:
**Describe the bug**
I used OpenTX companion to try to update my firmware on my Taranis X9D+ SE 2019 and after the update the radio wouldn't boot, etc. Latest vesion of Open Tx Compantion (2.3.9) used.

**To Reproduce**
Steps to reproduce the behavior:
Use OpenTX to update the firmware - it detects the radio type and downloads the x9d+ firmware.
Watch radio not boot.

WORKAROUND: Download firmware directly from FrSky's web site and use OpenTX Companion to upload. Radio works now, but I was not able to add the Lua script screen.

**Expected behavior**
The radio should have booted into the new firmware.

**Screenshots**
Sorry - couldn't get radio to boot to get screen shots.

**General information**
 - OpenTX version (e.g. 2.3.9)
 - Hardware (e.g. Frsky X9D+ SE 2019)
 - OpenTX Companion (e.g. 2.3.9 on MacOS Catalina)

3djc, 2020-09-18:
There is no detection, you have MANUALLY select the profile

closed 2020-09-18 by 3djc

querist, 2020-09-18:
Well, it still downloaded the x9d+ firmware because that was in the file
name and OpenTX companion was showing that it was downloading for the X9D+.

----------------------------
妇女能顶半边天
-- 毛泽东

3djc, 2020-09-18:
yes, because you DID NOT set the radio type to x9d+ 2019 in the radio profile, therefore the default was selected which is ... x9d+

querist, 2020-09-18:
Ah. Confusing interface. Thanks.

You may want to consider not having any default set so it won’t confuse
people new to this hobby.

----------------------------
妇女能顶半边天
-- 毛泽东

kilrah, 2020-09-19:
The first time you launch companion it opens the radio type/options selection dialog AND a popup window telling you to configure it for your radio. 

Don't know how to babysit users more.

querist, 2020-09-19:
In that case you may want to check the Mac version because it crashed the
first time I ran it and the second time it didn’t ask me to configure it.

----------------------------
妇女能顶半边天
-- 毛泽东