opentx/opentx #7949: OpenTx 2.3.9 -- Problem with "Adjust" function when applied to Global Variable of precision "0.0" and problem with "a>b" comparison for Global Variables and telemetry sensors with precision other "0.0"

aeronaut999, 2020-09-19:
Hardware: Taranis Plus (not 2019 edition)
OpenTX version -- 2.3.9 60e1edc1 , EEPR 219 (haven't checked for the same issues with older versions)
Companion version-- I'm not using Companion; I'm programming directly on the transmitter.

In OpenTx 2.3.9 I find that there is a problem with the "Adjust" function when applied to Global Variable of precision "0.0", and a problem with "a>b" comparisons for Global Variables and telemetry sensors with precision other "0.0"

The problem with the "Adjust" function is that Global Variables with precision "0.0" are "Adjusted" to values equal to one-tenth of what is actually commanded in the "Adjust" function.  This is true regardless of whether the Global Variable is being "Adjusted" to equal a numerical constant, or the value of a potentiometer knob, or the value of an "input", etc.

The problem with the "a>b" comparison is that Global Variables with a precision of "0.-" are interpreted as being tens times larger than their actual value.  In contrast, in "a>x" comparisons, the same Global Variables are interpreted at their actual face value.

The exact same thing is true for telemetry sensors of "0.-" -- telemetry sensors with a precision of "0.-" are interpreted by "a>b" comparisons as being tens times larger than their actual value.  

Also, telemetry sensors with a precision of "0.00" are interpreted by "a>b" comparisons as being one tenth of their actual face value.  

In contrast, in "a>x" comparisons, the same Global Variables and telemetry inputs are interpreted at their actual face value.

As result, the following program will fail to keep track of the maximum value of a telemetry input, variable knob, etc-- 

Program to keep track of the maximum value of potentiometer knob S2 -- 

Logical Switch 1 -- a>b -- S2-- GV1
Special Function 1 -- Logical Switch 1 -- Adjust GV1 -- S2

The program fails to work.  GV1 goes up as S2 increases, but then immediately goes right back down as S2 decreases.

Instead, we are forced to use this work-around:

input i10
input name -- i10
Source -- S2
Weight -- 10
Offset -- 0 (or minus 1 to make the program work a bit better by eliminating rounding errors and/or errors related to processing time)
Curve -- Expo --0
Switch -- ----
Side -- ----
Trim -- OFF

Logical Switch 1 -- a>b -- i -- GV1
Special Function 1 -- Logical Switch 1 -- Adjust GV1 -- S2

This work-around works regardless of whether GV1 has precision "0.-" or precision "0.0"

The same problem, solved by a similar work-around, arises when we are trying to write a program to keep track of the maximum value of a resettable altimeter -- that was how I first encountered this issue.  I couldn't compare my Global Variable to an input representing the value of the resettable altimeter.  Instead, I had to compare my Global Variable to a second input that was only one-tenth that size, and then whenever that comparison was "true", I adjusted my Global Variable to the value of the first input.

**_Even if only the problem with the "Adjust" function were fixed, the other problem with the "a>b" comparison could be easily lived with and the work-arounds described above would not not be needed, as long as the Global Variable in the comparison/ adjustment program were set to have precision "0.0"._**

I've documented these problem in much more detail in some posts on rc groups dot com.  The documentation includes several videos demonstrating the problems, and the complete text of the model programs used to demonstrate the problems and the work-around, as well as the complete text of the program used to successfully track the maximum value of a resettable altimeter.

The posts are --

[url]https://www.rcgroups.com/forums/showpost.php?p=45469301&postcount=3398[/url]
[url]https://www.rcgroups.com/forums/showpost.php?p=45469305&postcount=3399[/url]
[url]https://www.rcgroups.com/forums/showpost.php?p=45473383&postcount=3402[/url]
[url]https://www.rcgroups.com/forums/showpost.php?p=45474997&postcount=3404[/url]
[url]https://www.rcgroups.com/forums/showpost.php?p=45475001&postcount=3405[/url]
[url]https://www.rcgroups.com/forums/showpost.php?p=45475297&postcount=3406[/url]

aeronaut999, 2020-09-25:
In short the essence of the problem is that this simple program to keep track of the maximum value of potentiometer knob S2 cannot work, regardless of the "Precision Setting" of GV1 --

Logical Switch L1 -- a>b -- S2 -- GV1
Special Function SF1 -- L1 -- Adjust GV1 -- S2

Likewise, this simple program to keep track of the maximum value of alt3+ (up to 1000'), where alt3 is an altimeter that may be reset to zero in the middle of a flight (which also resets alt3+), also cannot work, regardless of the "Precision Setting" of GV1--

Input i1 -- source alt3+, scale 1000 ft, weight 100
Logical Switch L1-- a>b-- i1 -- GV1
Special Function SF1 -- L1 -- Adjust GV1-- i1

To see why these programs cannot work given the present architecture of OpenTX 2.3.9 and specifically the way that the "Precision Setting" of a Global Variable affects the way various functions operate, see the opening comment.

aeronaut999, 2020-09-26:
It appears that this would be one possible fix: 

> **_Why not restore some logic and consistency to the situation by allowing Global Variables as well as telemetry sensors to always function at their stated face value, regardless of "precision setting"?_** Or if the "precision setting" is really meant to reflect a "power setting", i.e. a "decimal multiplier", why not call it that? But even in that case, there ought to be _some_ setting for a given Global Variable that allows the simple comparison program given above to keep track of the maximum value of a telemetry sensor, input, variable knob, etc, to work properly. At present, that is not the case.

aeronaut999, 2020-09-27:
Hopefully the second post clearly shows the reader that there is a real problem here, even if it was obscured by the complexity and length of the first post.  

Presumably the problem has been present ever since the inception of OpenTX-- 

So, which of the two fixes suggested in the third post is the better one?

Note that changing the Precision Setting of a telemetry sensor does NOT change the numerical value of that sensor by an order magnitude.

So does it serve any useful purpose to have things set up so that when two different Global Variables with different Precision Settings get "Adjusted" to the value of some given switch output, input, etc, they end up being assigned values that are an order of magnitude apart?   As is presently the case.

If not, it would seem that first solution would be the better way to go.

Comments?

aeronaut999, 2021-01-16:
Still interested in seeing this fundamental flaw in the architecture of OpenTX corrected someday--

aeronaut999, 2021-07-20:
Still interested in seeing this issue fixed someday.

semininja, 2021-11-11:
A "drive-by" observation:

1. This issue suggests that there is a limit to the number of significant digits a variable can have, and the "precision" setting is a way to specify where they're placed. This makes sense to me. If this is the case, the min/max values allowed would change with the precision setting, and this could present a problem if the "adjusted" value exceeds the range allowed by the precision setting (overflow).

2. If the "allowed range" is changed by the "precision" setting, the "adjust" behavior is necessary to prevent overflow of the value. In this case, the comparisons must therefore be operating on the raw data (before the "precision" is accounted for). This is the part that would need changing; before comparing two values, the "magnitude" would need to be compared and compensated for in some way.

aeronaut999, 2022-02-23:
That observation seems to make sense, semininja.

aeronaut999, 2022-12-15:
I'd like to keep this issue open because it represents a fundamental flaw in the logic of OpenTX.