opentx/opentx #7952: Ideas-- calculated sensors to tell the bearing from start point (or xmitter) to current position of model, as well as direction model is currently travelling

aeronaut999, 2020-09-19:
Ideas-- calculated sensors to tell the bearing from start point (or xmitter location if GPS-enabled) to current position of model, as well as the direction model is currently travelling.  I would really enjoy having this information readily available as calculated sensors.

3djc, 2020-09-20:
Easy to do with LUA, also already transmitted by INav for exemple

Hobby4life, 2020-09-25:
Using OpenXsensor GPS already gives you de "Hdg" output.. Bearing of plane.

aeronaut999, 2020-09-25:
Good to know, but I don't fly w/ that sensor; FrSky GPS sensor does not give "Hdg" output.

Hobby4life, 2020-09-25:
> Good to know, but I don't fly w/ that hardware; FrSky GPS sensor does not give "Hdg" output.

I can say, start using them.. they perform far more better then the original GPS'ses :-)

Faster fix due availability of Better performing GPS'ses like Ublox Neo8> chipsets.
Faster Position updates Frsky = 1x /sec, Ublox = 10x /sec (handy for ground speed measurement)
GPS Altitude very accurate, due the fact Ublox uses Multiple Sattellite platforms for calculation.
Gives status output, Sats in view, Hdop etc..
Costs a lot less :-)

bigtwin, 2021-02-11:
Eventhough I would also love to have this feature, I dont think this is something which can be done in a transmitter in a sensefull manner. I know from other GPS sensors that they are providing those values. Just as OpenXSensor. Its a "problem" of the FrSky Sensor and I dont think this can be addressed within OTX.

lshems, 2021-02-11:
Of course it could. But Lua is probably where the Devs will point.

No reason why it would not be possible.

closed 2022-04-17 by stale[bot]