opentx/opentx #7955: OpenTx 2.3.9 - Cannot rename log file in radio

VancouverUmbrella, 2020-09-22:
**Equipment**
OpenTx 2.3.9 (and previous versions IIRC)
FrSky QX7 (legacy) or QX7 ACCESS

**Description**
It seems impossible to rename a log file in the radio. The file name simply reverts to its original name.

**To Reproduce**
1. Navigate to the logs folder on the radio.
2. Long click on a log file and select, 'Rename'.
3. Edit the file name e.g. change the first letter of the name.
4. Click the 'Exit' button on the radio to apply the change.
5. File name reverts to the original.

**Expected behavior**
The edited file name would persist.

**Question**
Am I possibly making a mistake by pressing the 'Exit' button to apply the change? I've tried every other possibility I can think of but none work.

mpjf01, 2020-09-22:
The log file should always reflect what is happening - that's the point of it. 

When telemetry sensors are altered or added the log file ceases to be an accurate record of events until the next day's file is created. See #7752 . The preferred solution at present has been to rename the existing file when a change is made and a new file will be automatically created. As this doesn't work the user has to either put up with a potentially corrupt log file that does not show what happened or delete what might be very useful information in order to get a new log file created. 

Surely the soundest option is to automatically create a new log file whenever a sensor change occurs. Manually renaming the old file is not intuitive - most users don't even know that the old log file is likely to be useless when sensors are changed, and hardly anybody is going to remember to rename it. Renaming files manually to get around this is not really the appropriate way to deal with it.

VancouverUmbrella, 2020-09-22:
Yes, I see the point about sensors being changed. 

However, my use case is different. I often test various different configurations of receivers or antenna placement on a given model on a given day without changing sensors. Renaming the log file for each different configuration makes keeping track of the data much easier. As things stand, I must connect the radio to a PC to accomplish the renaming.

3djc, 2020-09-22:
I think you just have to go to the end of the file name

VancouverUmbrella, 2020-09-22:
Thank you, Yes. 

You can make your changes, go all the way past the file name to a blank space, then do a long press on the selector button. (Info posted on RCGroups by @mikeshellim and @pmackenzie.)

I wouldn't have guessed that method but it does work perfectly. 

Perhaps the UX could be made more consistent with other functions that allow a press on the Exit button to commit edits. Or am I mistaken in thinking that? In any case, I think it's fair to close this bug now.

Thanks again for the very quick response.

Best,
Andrew

closed 2020-09-22 by VancouverUmbrella

kilrah, 2020-09-22:
The rename function has always been a bit broken, would be worth reworking it completely one day....