opentx/opentx #7958: Only one "Serial port (TTL)" Displayed on Hardware menu.

al21883, 2020-09-22:
On the Haware screen (System menu) there's only one "Serial port (TTL)" available.
It's a brand new radio, and i haven't done any HW modification. I have reflashed the firmware to the latest (Stable) available, and the problem persist.

Thanks in advance for your help!

 - OpenTX version: 2.3.9-otx (Crossfire, faichoice, flexr9m, internalmulti, multimodule, lua, luac, ppmus)
 - Radiomaster TX16S
 - OpenTX Companion 2.3

3djc, 2020-09-22:
Clearly not here

[image]

[image]

al21883, 2020-09-22:
I don't understand. Same version, same Hardware:

[image]
[image]

3djc, 2020-09-22:
You have BT option

closed 2020-09-22 by 3djc