opentx/opentx #7961: SC and SH problem

Flierman1945, 2020-09-23:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - OpenTX version (e.g. 2.1.23 or N123)
 - Hardware (e.g. Frsky X9D+ or 9X Pro)
 - OpenTX Companion (e.g. 2.1.23 on Ubuntu 16.04)

Flierman1945, 2020-09-23:
When selecting Simulate SC jumps to the middle position.  It should open in SC up.

kilrah, 2020-09-23:
No info, template is meant to be filled.

Flierman1945, 2020-09-24:
Sorry, I ment to continue, but could not find my way around on how to post, so my message was unfinished.  But I just posted this on RC forum and think it might give you an idea of my problem and perhaps offer advice.

 have a Tx16s and am having some problems with Companion 2.3.9. When I open companion and select any model from my model list, open it and then select Simulate, SC is always at mid pos, speech file says engine armed ( SF down), but SF is up and trainer switch ( SH) is on. Yet none of them are active. Even if I generate a new model it is the same. Transferring my model list over to my win10 laptop and following the same proceedure, there is no problem, SC stays in the up position SF voice says engine off ( not engine on) and SH voice is saying trainer off.
I have reinstalled several times, even tried a nightly on my win10 desktop, but it is still the same.
So I think there must be another few files burried away, say in temp folders and this could be the problem.
I run another version 2.2.4 along side and this works correctly. I am positive now there is a problem on my desktop, but how to fix it is the question. An interesting point is when the models are wtitten to the Radiomaster, they operate correctly and as expected.
I bevlieve there is also an issue with the P1-P6 rotary switch used in Companion, but when you upload to the tx, that works correctly.

kilrah, 2020-09-24:
[image]

closed 2020-09-24 by kilrah

Flierman1945, 2020-09-24:
Thx for the solution