opentx/opentx #7965: Reverbed Audio X12S

Hobby4life, 2020-09-24:
**Describe the bug**
Occasionally Reverbed audio on X12S, I dont know if this is caused by OpenTx.

**To Reproduce**
Steps to reproduce the behavior:
Occasionally there is Reverbed audio on my X12S, directly from turning on the radio, switching it back off and on again and Audio is normal non reverbed.

**Expected behavior**
Normal audio :-) no Reverbed although it is not disturbing

**Screenshots**
[url]https://youtu.be/p-ALz_kaiJo[/url]

**General information**
 - OpenTX version (e.g. 2.3.9)
 - Hardware: Frsky Horus X12S Access upgrade
 - OpenTX Companion v2.3.9

kilrah, 2020-09-24:
Official sound pack or 3rd party?

Hobby4life, 2020-09-24:
Good point, I am now using TTSautomated files.

I will try official soundpack, and let you know

closed 2021-06-26 by stale[bot]