opentx/opentx #7966: X9D+ SE, Flashing Multiprotocol Module using context menu

fiechr, 2020-09-25:
**Describe the bug**
Firmware files (like `multi-stm-opentx-aetr-inv-v1.3.1.69.bin`) don't show up on the SD card, while FrSky firmware files with long filenames (like `RXSR_ACCST_2.1.0_Fport_FCC.frk`) work fine.

**To Reproduce**
Copy firmware file(s) onto SD card using USB connection. Target is the subfolder `/FIRMWARE/MPM/`.

**Expected behavior**
Files should show up and be "flashable" using the context menu. The context menu should show "Flash external module".

Renaming files to **short** filename does make them show up. Not tested, but I guess only DOS-like 8.3 filenames seem to work.

**General information**
 - OpenTX version 2.3.9
 - FrSky Taranis X9D SE (pre 2019)
 - Linux Kernel 4.15 on Manjaro Linux, to copy the files to the SD card

3djc, 2020-09-25:
Max size depends on LCD size, by design

closed 2020-09-25 by 3djc

fiechr, 2020-09-25:
So why are other files, like for example text files, shown with the full length filename?

3djc, 2020-09-25:
RXSR_ACCST_2.1.0_Fport_FCC.frk is 30 chars
multi-stm-opentx-aetr-inv-v1.3.1.69.bin  is 39 for a max of 32 for your radio

fiechr, 2020-09-25:
Thank you!

In my personal opinion, I think, it would be better to show "something" instead of nothing, because people will be confused. On the other hand, I can not estimate how much work that would be to implement and if it would be really worth the time.