opentx/opentx #7967: Channel monitor graphical/display glitches

xnoreq, 2020-09-25:
**Describe the bug**
The channel monitor has weird graphical/display glitches where certain columns have a different color/contrast.

See photo:
[image]

**General information**
OpenTX 2.3.9 on X9-Lite S

3djc, 2020-09-25:
Fixed already

closed 2020-09-25 by 3djc

xnoreq, 2020-09-25:
With which commit/version?

xnoreq, 2020-09-25:
Still the case in 2.3.10N330 with sdcard-128x64-2.3V0032.

3djc, 2020-09-25:
Do you still have the empty rectangle or just a bar ?

3djc, 2020-09-25:
Ok so x9lite also has the crappy screen, will do like x7

3djc, 2020-09-25:
You should be able to test with tomorow 's nightly

xnoreq, 2020-09-25:
Thanks. https://github.com/opentx/opentx/commit/3715669b9155ec8abac703252c90f1b8644b87c2