opentx/opentx #7970: Half of the 6 Position Switch weight values assigned are incorrect

jims1234, 2020-09-26:
If you select System/ Hardware Calibration/ then Analogs Test.  Then look at the 6 Position switch levels shown as you select device “06”, the 3 negative values are Ok but the three positive ones are incorrect, causing some ambiguity during Calibration
Specifically:
The switch displays 19 instead of 20 when 6pos switch position 4 is selected, then 59 and finally 99 as positions 5 and 6 are selected instead of 60 and 100 on the positive side.
Recommend correcting these three positive side values in an OpenTx 2.3.9 firmware update to read 20, 60, and 100 so users calibrating the switch can do so with better accuracy and less confusion.  Positions 1 to 6 should be set to values: -100, -60, -20, 20, 60, and 100 respectively. 

If you refer to this YT video immediately below at approximately 24:00 minutes in, it clearly describes, and shows the issue and how to check for it.

https://youtu.be/L-x1YaWL45Q
 
**General information**
 - OpenTX version 2.3.9 and possibly some earlier radios.. not sure of the scope.
 - Hardware (all Radiomaster TX16S, Jumper T16, and T18)
 - OpenTX Companion (Windows 10 2.3.9) and possibly others

Mateyhv, 2020-10-12:
I dont know if thas OTX issue or the button signal. You can solve it in several ways but its definitely not a problem if you want to use those switches. Its more importan that they remain steady what they do.

jims1234, 2021-04-30:
Mateyhv,
Sorry I missed this reply till just now.. I don't know either but this appears to be an openTX issue that degraded a users ability to calibrate his Radio since the three positive values he will calibrate sticks to will be slightly OFF.. at Time 26:00 in the video you'll hear that it has annoyed several users and the impact on them..  I am not sure it it applies to the older Taranis Radios or not but seems to impact all the newer ones with color LCD's..

closed 2022-04-16 by stale[bot]