opentx/opentx #7971: Compilation leads to boot loop

xnoreq, 2020-09-26:
**Describe the bug**
Building 2.3 on my own results in a bin that when flashed causes the TX to bootloop 

**General information**
 - Hardware: x9lites
 - arm-none-eabi-gcc 10.2.0
 - arm-none-eabi-newlib 3.3.0
 - arm-none-eabi-binutils 2.35
 - cmake 3.18.3

**To Reproduce**
Here's what I did:
1. Cloned the repo
2. Installed above packages
3. In my build dir I ran: 
```
cmake ../opentx/ -D DISABLE_COMPANION=1
../opentx/radio/util/build-firmware.py firmware $(pwd)/opentx-x9lites-lua-noheli-en.bin
```
4. The result is a 418K big bin
5. Flashed bin on my TX
6. Boot loop

In order for it to compile I had to fixed the following lines in
radio/src/strhelpers.cpp:
```
diff --git a/radio/src/strhelpers.cpp b/radio/src/strhelpers.cpp
index 774fe52..20f877f 100644
--- a/radio/src/strhelpers.cpp
+++ b/radio/src/strhelpers.cpp
@@ -218,7 +218,7 @@ char * getTimerString(char * dest, int32_t tme, uint8_t hours)
     *s = '\0';
   }
   else if (tme < secondsPerYear) {
-    qr = div((int) tme, secondsPerDay);
+    qr = div((int)tme, (int)secondsPerDay);
     div_t qr2 = div(qr.rem, 60);
     *s++ = '0' + (qr.quot / 100);
     *s++ = '0' + (qr.quot / 10);
@@ -230,8 +230,8 @@ char * getTimerString(char * dest, int32_t tme, uint8_t hours)
     *s = '\0';
   }
   else {
-    qr = div((int) tme, secondsPerYear);
-    div_t qr2 = div(qr.rem, secondsPerDay);
+    qr = div((int)tme, (int)secondsPerYear);
+    div_t qr2 = div(qr.rem, (int)secondsPerDay);
     *s++ = '0' + (qr.quot / 10);
     *s++ = '0' + (qr.quot % 10);
     *s++ = 'Y';
--
2.28.0

```

Otherwise it would complain about the call to `div` being ambiguous.

The build log:
```
cmake
-DPCB=X9LITES
-DHELI=NO
-DPPM_UNIT=PERCENT_PREC1
-DLUA=YES
-DGVARS=YES
-DOVERRIDE_CHANNEL_FUNCTION=YES
-DMODULE_PROTOCOL_D8=YES
-DAFHDS3=NO
-DTRANSLATIONS=EN
/home/user/dev/opentx/opentx/radio/util/../..-- OpenTX 2.3.10 @a0ce3f5d
-- Python found, version: 3.8.5
-- Googletest will be downloaded
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xnor/dev/opentx/build
CMake Deprecation Warning at CMakeLists.txt:14 (cmake_policy):
  The OLD behavior for policy CMP0023 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

TARGET simu/libsimulator: cpp compiler /usr/bin/c++ v10.2.0
CMake Warning at radio/src/tests/CMakeLists.txt:63 (message):
  WARNING: gtests target will not be available (check that GTEST_INCDIR,
  GTEST_SRCDIR, and Qt5Widgets are configured).

TARGET bootloader: cpp compiler arm-none-eabi-g++ v10.2.0
TARGET firmware: cpp compiler arm-none-eabi-g++ v10.2.0
[  5%] Built target 9x_xbm_1bit
[  5%] Built target lua_export_x9lites
[  5%] Built target firmware_translations
[ 18%] Built target 9x_fonts_1bit
[ 19%] Built target 9x_bitmaps
Scanning dependencies of target bootloader
[ 21%] Building CXX object radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/__/__/__/__/__/stamp.cpp.o
[ 21%] Building CXX object radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/__/__/__/__/taranis/bootloader/boot_menu.cpp.o
[ 21%] Building CXX object radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/__/usbd_storage_msd.cpp.o
[ 21%] Building CXX object radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/boot.cpp.o
[ 22%] Building CXX object radio/src/targets/common/arm/stm32/bootloader/CMakeFiles/bootloader.dir/bin_files.cpp.o
[ 22%] Linking CXX executable bootloader.elf
[ 32%] Built target bootloader
Scanning dependencies of target firmware
[ 32%] Building CXX object radio/src/CMakeFiles/firmware.dir/debug.cpp.o
[ 32%] Building CXX object radio/src/CMakeFiles/firmware.dir/stamp.cpp.o
[ 32%] Building CXX object radio/src/CMakeFiles/firmware.dir/lua/api_general.cpp.o
[ 32%] Building CXX object radio/src/CMakeFiles/firmware.dir/targets/common/arm/loadboot.cpp.o
[ 32%] Building CXX object radio/src/CMakeFiles/firmware.dir/targets/common/arm/stm32/usbd_storage_msd.cpp.o
[ 32%] Linking CXX executable ../../firmware.elf
[100%] Built target firmware
```

kilrah, 2020-09-26:
You must use the compiler version mentioned in the wiki.

closed 2020-09-26 by kilrah

xnoreq, 2020-09-27:
If I want to get started on opentx development and wanted to figure out why it is bootlooping do you have hints or pointers as to what is wrong? Thanks.

kilrah, 2020-09-27:
I think some stuff has been done in the 2.4 branch to support more recent compilers already, but I haven't been following that myself

xnoreq, 2020-10-14:
Can we re-open this issue? I would be very interested in the changes required to make opentx bootable when compiled with more recent compilers.
Maybe someone who has followed this development in the 2.4 branch can point out the commits?

3djc, 2020-10-14:
2.4 will use 6.7