opentx/opentx #7974: FrSky R9M 2019 JR modul increase buzzing as the power output increases

dusan164, 2020-09-27:
Hello for the first time.
I am from Slovenia and I do not know if this is the right place but I will try.
I am using OpenTX 2.3.9 latest stable version on my FrSky Horus X12S radio. All worked OK before using external FrSky JR R9M 2019 modul with R9M_FLEX firmware.
After reading a few posts with no resolution to the problem, I am asking you for an answer about buzzing sound: modul increase buzzing as the power output increases (can not use other sound notifications on the radio).
The only way to stop it is set volume to zero (but it is not an option as I need other notifications on the radio).
I tried with FrOS FW-X12S-V1.6.06 and there was no buzzing at all.
Is this something related to OpenTX app, X12S hardware issue or what?
If you can give me some advice I will appreciate that a lot. Someone post that he used foil tape to shield the speaker/amplifier board in the radio.
Thanks in advance.

Hobby4life, 2020-09-30:
This is a hardware technical problem.. not related to OpenTX.

dusan164, 2020-09-30:
Thanx

brainbubblersbest, 2020-11-02:
> This is a hardware technical problem.. not related to OpenTX.

It seems fros handle the TX internals in another way.
As described in the duplicate

dusan164, 2020-11-03:
Tnx.

FLYN-001, 2021-05-21:
Own a X12S here too. The noise is annoying on internal 2.4 & crazy annoying on TBS CRSF. 
I have reduced it quite a lot with grounded shielding on the speaker wires & speakers. Also grounded shielding on the inside of the rear case for the module bay. 

But the thing that helped the most was extending the speaker wires & adding a ferrite ring filter. 

An XJT module in the rear bay is now silent. ACCST internal is only just audible. 
ACCESS internal is slightly louder than ACCST but still 70-80% quieter than it was as standard. 

I would very much like a resolution to this problem that might shut down the amp when no audio is playing. Donations would flow!!