opentx/opentx #7975: Sounds drop down menu issue Taranis X9D+SE 2019

Martivip, 2020-09-28:
Hello 

**Describe the bug**
When in SOUND/en folder there is more items than 512, then sounds list on dropdown menu is not properly displayed

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Insert 513 or more items (sounds files) on SD card SOUNDS/en folder
1. Go to Special Functions
2. Click on SF1
3. Choose Play Track
4. Choose sound file - and right now you will see that dropdown menu with list of the files is corrupted

When i restore maximum number of 512 files in SOUNDS/en folder everything comes back to normal.

**General information**
 - OpenTX version 2.3.9 (60e1edc1)
 - Hardware Taranis X9D+ SE 2019
 - FW Version OpenTX-x9d+2019
 - OpenTX Companion 2.3.9

3djc, 2020-09-28:
How was your SD card formated ? Would be interesting to try with it formated as FAT32 (NOT exFAT)

Martivip, 2020-09-28:
my SD card is formatted to FAT32
[image]

Martivip, 2020-10-06:
does anyone is able to confirm this problem ?

Martivip, 2020-10-06:
after OpenTX update to 2.3.10 the issue still exists

mhon44, 2020-10-09:
I had the same problem on my X-Lite, OTX 2.3.9, also on 2.3.10. It seems like the LUA Script was not updated recently. 

Solved it simply by deleting corrupted .waw files

Martivip, 2020-10-09:
> I had the same problem on my X-Lite, OTX 2.3.9, also on 2.3.10. It seems like the LUA Script was not updated recently.
> 
> Solved it simply by deleting corrupted .waw files

I don't think that issue is related to corrupted .wav files, please do this exercise :
when you delete this BF .wav files, please replace them by 5 different ones and the issue will come back :(

mhon44, 2020-10-09:
Actually that is true, but I am OK with deleting few files - there are hundreds of tracks I would never use.

Looks like there is a limited number of .waw files in a folder that OTX can show with no issue

closed 2021-09-04 by stale[bot]