opentx/opentx #7976: 6 position switch on EX1 or EX2 input on Radiomaster TX16S

Tygrys-1, 2020-09-28:
One can set EX1 or EX2 as 6 position switch on the hardware page. But they do not act the same as 6P. S1 and S2.

When 6P, S1 or S2 is set to a 6 position switch, in switch menus (in any place to select switches as conditions) there will be 6P1 .. 6P6 or S11 .. S16 option to choose as a switch condition. For example: a switch to activate a flight mode on modes page of the model.

This however is not the case when EX1 or EX2 are set as 6 position switch on the hardware page. They show only as EX1 or EX2 choice, without positions specified. There is no EX11, EX12 and so. So - what is the point of setting them to 6 position switch on hardware page?

I'd like to add 6 position switch on EX1, so I'd like it to act like a 6P or S1/S2 set as 6 position switch. 

Can this be implemented somehow?
It might be a problem with length limitation of the switch name, because EX11 will be 4 letters, while others are max 3.
Maybe need to rename EX1 to E1.

3djc, 2020-09-28:
Had a quick look at the code, I see nothing obvious preventing it. Did you move them correctly during calibration ?

Tygrys-1, 2020-09-28:
That is maybe not :-) Thank you.

closed 2020-09-28 by Tygrys-1