opentx/opentx #7978: LuaC taranis x9d+ 2019 - yaapu

babolcs, 2020-09-29:
I installed the yaapu script on my taranis x9d+2019. On the telemetry screen i can see the sensors, but on the yaapu screen i can see no telemetry data. 

At firmware installation for this model there is no luac option just lua. Can it be the problem?

3djc, 2020-09-29:
This site is only for OpenTX bug tracking, that your issue really isn't. Please go to community sites like RCgroups or openrxforums for community support

closed 2020-09-29 by 3djc

babolcs, 2020-09-29:
Thanks a lot for your help. 
Yes i was skeptic  to post it here, but didnt know where to.

3djc, 2020-09-29:
Yaapu does have a specific thread on rgc : https://www.rcgroups.com/forums/showthread.php?3020527-An-Ardupilot-frsky-telemetry-LUA-script-for-the-Horus-X10-X12-and-Taranis-X9D-E-QX7

babolcs, 2020-09-29:
Sorry one question.
Is it normal that there is no luac option for taranis x9d+ 2019?
Thx

3djc, 2020-09-29:
Yes, luac has been defaulted in all build for a little bit over a year

babolcs, 2020-09-29:
Really thanks a lot for your help. Have a good day. Szabolcs