opentx/opentx #7979: TX16s USUALLY not recognized as Joystick

Soundserfer, 2020-09-30:
Opentx 2.3.9
MacOS 10.15.6
Companion 2.3
Radiomaster TX16s controller
USB-c to USB-c cable (verified as a working cable)

Note: this has worked twice, and the controller could then be used in computer simulators (FPV drones)

Procedure
Companion > Preferences > Simulator Settings > Joysticks > Enable checked : "No joysticks found" appears in the controller window, greyed out. 

On the 2 occasions that it worked, the window read "Radiomaster TX16s", not greyed out, and simulators worked as expected

I have tried all combinations of starting 1) Radio, 2) Simulator, 3) Companion 2.3 in various sequences.

kilrah, 2020-09-30:
You probably don't have the USB mode set to joystick on the radio.

Soundserfer, 2020-09-30:
But I do

3djc, 2020-09-30:
I would think something on your side creating that. You can rule out simulator or companion, there must be a way on mac to see when a joystick is connected, it is all that matter (sorry, not a mac user :)

kilrah, 2020-09-30:
Did you actually set it to Joystick in settings? People often confuse the 2 options on the popup and choose the wrong one.

Soundserfer, 2020-10-03:
Yes, as in my post, ASK had been default in USB, and I changed it to Joystick

and0x000, 2020-10-06:
Same here, I found out, that Windows might assign the incorrect driver. Please check the device manager, if there is a device under 'libusb-win32 devices' called 'BETTER_USB_HS'. This is most likely your TX16s (at least with me, this occaisonally is the case). I need to reassign the human interface device driver instead. Right-click the device, select 'properties', 'driver' and then 'rollback driver'.

If the 'rollback driver' option is not available, select 'update driver' instead. Go to 'search on my computer', 'select from list of available drivers'. There you should see the USB input device driver available. Select this and confirm the change.

The radio should now become recognized in simulators such as Liftoff and the likes again.

This is only a transcript from [this video](https://www.youtube.com/watch?v=SDnRSL3rDwM). All credits go to it.

kilrah, 2020-10-06:
He said he's running mac os, not Windows.

Soundserfer, 2020-10-11:
I have installed the simulator on a different Mac with an earlier OS (10.14 rather than 10.15), and the TX16s is recognized every time.

BTW, 10.15 has discontinued support for 32 bit apps; now 64 bit only. 
But to repeat myself, 10.15 SOMETIMES works.
A BetaFPV controller I have is recognized as Taranis, and works on both platforms every time.

0x00000002, 2021-06-15:
Hey, still have the same problem. isn't recognised by macos.

charlesmims, 2022-07-08:
Commenting to hopefully keep this active. I'm experiencing this as well. New M1 MBP, Monterrey, TX16S not detected.

jstelten, 2022-09-03:
Experience the same with mac ventura

palaniraja, 2022-09-20:
I had the same problem when I used a USB-C to USB-C cable with radio and mac, but using USB-C to USB-A  comes with radio itself with a USB-A adapter seems to detect and present the option to pick as joystick

Monterey 12.6 on intel

doHernandezM, 2022-11-10:
Ditto:
--------------------------------------------------------------------------------
M1 MBP 13" w/ macOS Ventura
FrSKy Taranis X9 Lite S

• I can unplug the MBP and plug the transmitter into a macOS Monterey Mac(with native USB 2) or a Dell w/Windows 11 and OpenTC Companion/the OS see it just fine.
• Tried various powered and unpowered USB C -> USB A (3/SS) and/or USB 3 - USB 2, just a basic usb c cable to transmitter.
• SD card works but Joystick/bootloader does not. 
• I'm savvy and willing to help troubleshoot.

dfu-util output:
--------------------------------------------------------------------------------
sudo dfu-util --list -v -v -v
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

libusb version 1.0.26 (11724)
[timestamp] [threadID] facility level [function call] <message>

[ 0.004109] [0000a706] libusb: debug [libusb_get_device_list]  
[ 0.004137] [0000a706] libusb: debug [libusb_get_device_descriptor]  
[ 0.004142] [0000a706] libusb: debug [libusb_get_config_descriptor] index 0
[ 0.004162] [0000a706] libusb: debug [libusb_exit]  
[ 0.004188] [0000a70a] libusb: debug [darwin_event_thread_main] darwin event thread exiting
[ 0.004304] [0000a706] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.004592] [0000a706] libusb: debug [usbi_remove_event_source] remove fd 3

System Profiler info:
--------------------------------------------------------------------------------
USB 3.1 BUS
v USB 3.1 BUS
    FrSky Taranis Bootloader

FrSky Taranis Bootloader:

  Product ID:	0x5720
  Vendor ID:	0x0483  (STMicroelectronics)
  Version:	2.00
  Serial Number:	00000000001B
  Speed:	Up to 12 Mb/s
  Manufacturer:	OpenTX
  Location ID:	0x01100000 / 1
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0
  Media:
     ...SD CARD STUFF.....

rostyslavio, 2023-09-21:
I have TX12 controller and same issue with connection
It not recognized my on MacBook Pro M1 2020
But it ok on MacBook Pro 2018 Intel and on MacBook Pro M1 **Pro** 2021 with TX original cable + adapter from usb to usb-c (also tried with Apple original cable and not worked)

Clasko, 2023-11-29:
I can confirm that USB-C -> USB-C on Apple Silicon does not work. USB-A + Adapter -> USB-C works.

upya4ko, 2024-11-05:
Same problem (WinXP inside vBox) Jumper **t12pro**, in system **USB\VID_0483&PID_5720\** - in **DFU** mode, driver from SDCARD folder **usb_driver** appears as after first boot "USB Mass Storage Device" installed using ZliDag (i try make it workin on debian, but it work only on debian 9 so it difficult, so i use VM) 
 
When experimenting i flash some bootloader from Transis it create folder with files **WinUSB Generic Device** and **STM32  BOOTLOADER** **VID_80EE&PID_0021**

In **USB Joystik mode** in linux it appears as ```1209:4f54 Generic FrSky Taranis Joystick``` **HID\Vid_1209&Pid_4f54&Rev_0200**  WinXp is see ```FrSky Transis Joystik```   switches appears as **sliders**  AUX 4 in center position is bitton  1, work wrong completely.
i try all drivers from SDCARD not connect, i think problem in wrong driver, 

As i see latest updated companion2 app for MAC i think main DEV team on MACs, i hope someone make ```.Appimage``` for linux with old libs (i not have expierience in this) 

it make install on any system without dependencies

hope it may UP problem and find solution.