opentx/opentx #7980: X12S with OTX2.3.9 - volume reduce issue

MWunderer, 2020-09-30:
Description:
I have OTX 2.3.9 on my Horus X12s. When i set to play two or more sounds on same trigger (switch, LS, value etc.) the first one will play normal, the second one are played with a much reduced volume, nearly not to hear. After played both, Volume becomes normal. Next time it plays a sound, volume is as set before. Sounds are made with TTSAutomate but normal. If i trigger them individual, they work perfectly. It also has the same issue reading values. If i set a switch (SH->) to read both timer1 and timer2, first value is read with normal volume, second timer i can't properly hear. Also longer sounds the same, too. If i play a sound with two longer words and a gap in between, first word will be spoken normal, afterwards the volume drops and the rest of the soundfile is hardly to hear.

To Reproduce:
Set in the special functions two lines. Both lines as trigger same switch, action is 'play sound', then set up which sound to play. 
Trigger the action. First line will execute normal, secodn one with reduced volume

important hint: i have set in global functions the S2 as volume control. All inputs are well calibrated and volume control is working properly

Expected behavior
Maybe the code will use while caling first sound file the right volume value from variable. After finishing that, the variable of sound volume will be reseted to a default value and next play use this value. Leaving the routine, afterwards the system rewrite the variable with the coccrect value and next time playing single sound it's ok. 

- OpenTX version 2.3.9)
- Hardware Horus X12S

Hobby4life, 2020-09-30:
I will test this too tonight.

Did you by chance ever noticed any echo/reverb audio from your x12s?

MWunderer, 2020-09-30:
Hello
Nope. There was no echo or reverb i notice. But i tried now a bit and i can reproduce it also in companion. Even there it does the same thing. Set special function on SH to read timer1 and next row same switch to read timer2. Once i activate the switch it reas both timers with same volume, next time it starts after the first to reduce volume, next time already on half on first timer, next on half of second timer and so on. A bit random.
Also i found out that it has to do with the volume control via S2. When i disbale globale function 'volume set' and set the volume only via the main settings, it works quite normal. Then also following sounds or values a read with same volume. In my opinion its 99% a bug somewhere in the routine from where it takes the volume value

3djc, 2020-09-30:
Can you post your otx ?

MWunderer, 2020-09-30:
[HorusX12_Standard_V1.zip](https://github.com/opentx/opentx/files/5306510/HorusX12_Standard_V1.zip)

MWunderer, 2020-09-30:
This is the version which uses standard sounds and volume control by S2. If you activate SA- it plays three sounds. First will be expressed normal, second reduced and last one you can rearly hear..

3djc, 2020-09-30:
...

Instead of trying to guess what bugs we have, why don't you look first at your own setup ? You global function for volume is triggered by Act, which is only true for 3 seconds or so after last switch move. Replace it by ON ...

closed 2020-09-30 by 3djc