opentx/opentx #7983: X9D+2019 R9M-2019 ACCESS V1.3 It's not responding

mimicookie, 2020-10-02:
"OpenTX 2.3.10-N355" farm cannot set the output power of the R9M-2019 (ACCESS V1.3) module.

The information screen of the module does not appear.

It works properly in OpenTx2.3.9.

 - OpenTX 2.3.10-N355
 - Hardware X9D+2019SE + R9M-2019(V1.3 Flex)
[image]

3djc, 2020-10-02:
Thaks for reporting it. I'll prepare 2 or 3 binaries for you to test to try to narrow down when this issue appeared

mimicookie, 2020-10-02:
Thank you for your help.

3djc, 2020-10-02:
Could you try this one. Just install from SD, just got to module version and just check if showing or not
[firmware-test1.zip](https://github.com/opentx/opentx/files/5316342/firmware-test1.zip)

mimicookie, 2020-10-03:
Nothing has been improved.
The module version is not shown either.

kilrah, 2020-10-03:
You did not say if control works properly or not.

3djc, 2020-10-03:
OK, right now just trying to narrow down when it broke, making you another test version

3djc, 2020-10-03:
Could you try this
[firmware_test2.zip](https://github.com/opentx/opentx/files/5321392/firmware_test2.zip)

mimicookie, 2020-10-03:
Same as "test1". The module version is not shown. The module version is not displayed, and the transmission power cannot be set.

3djc, 2020-10-03:
Could you try this one ?
[firmware_test3.zip](https://github.com/opentx/opentx/files/5321409/firmware_test3.zip)

mimicookie, 2020-10-03:
"test3" works fine.
It also displays the module version. RF output can be set up by option.
It's perfect.

3djc, 2020-10-03:
Tahnks a lot, it confirms that current issue is linked to FRM302 PR, we now need to find the issue there then

mimicookie, 2020-10-03:
Thanks for the effort.

3djc, 2020-10-03:
Could you test this ?
[firmware_testfix1.zip](https://github.com/opentx/opentx/files/5321625/firmware_testfix1.zip)

mimicookie, 2020-10-03:
"firmware_testfix1" works perfectly.
No problem.

3djc, 2020-10-03:
GREAT NEWS !

PR #7987 (2020-10-04): Fix X9D 2019
This fixes #7983
closed 2020-10-04 by 3djc