opentx/opentx #7985: Error after flashing firmware to x9d+SE2019?

pau33, 2020-10-02:
A very strange behavior.

I bought a new x9d+SE2019.
With companion 239 I downloaded FW:
opentx-x9d+2019-lua-noheli-en-2.3.9-otx.bin and opentx-x9d+2019-flexr9m-lua-noheli-en-2.3.9-otx

I flashed it to the radio. When I hit exit in bootloader mode, the screen turned of,  but the Power LED didn't switch of. Instead it glows light. There is no reaction on the power button. No way to switch on and no way to got to bootloader mode. Nearly a brick.

I then disconnected the  battery (2s Lipo, 3.87V/cell) and connected it again after some time. LED is not glowing. I still couldn't turn the radio, but I could got to bootloader mode.

I then downloaded the 239 Firmware on the FrSky Homepage:
firmware_x9d+2019_en_200617.bin
and flashed it.

The radio started up normally.
Then I updated the bootloader to 239 an tried the procedure above:
It is the same: It does not work with opentx-239 (both ones above) but it works with FrSky-239 Firmware.

Has anybody seen this behavior? May it be a hardware problem? Or is something wrong with the opentx firmware for x9d+SE2019? Or have I done something wrong?

Thank You in advance.

[image]
[image]
[image]

kilrah, 2020-10-02:
Which flashing method did you use? Put the file on the card in the FIRMWARE folder and use the "write firmware" bootloader function.

pau33, 2020-10-02:
I flashed the frsky 219 firmware via companion to the radio.
Afterwards I copied it to the Firmware folder on the radio and flashed the bootloader with "write firmware" via the radio, just as you mentioned it (with the frsky firmware). Originally it was the 230 bootloader.

But flashing the opentx219 firmware after that again was not successfull (as above).

3djc, 2020-10-03:
Flashing from companion is sometimes not working, and this radio dis very sensitive to it. Flash the OpenTX release  from SDcard like you did the FrSky one

3djc, 2020-10-03:
Duplicate #7742

closed 2020-10-03 by 3djc

pau33, 2020-10-03:
I reopen this point only for information. 

Writing opentx firmware vie internal Bootloader (shorten the name if too long) worked at once.

To sum up:
- Writing FrSky 239 FW via Companion worked at once several times.
- Writing opentx 239 via Companion never worked.
- Writing opentx 239 with internal Bootloader worked at once.

It seems, as if FrSky modified something in their 239 FW to make it more robust for flashing with Companion and Windows10. May be the tell you?

Thanks for your help again.