opentx/opentx #7993: Frsky X9D+ SE 2019 and R9M 2019 with R9MX RX can't register

uragano47812, 2020-10-06:
I have a Frsky X9D+ SE 2019 with a new R9M2019 and a R9MX Access RX. I'm running last version of opentx 2.3.9. All firmwares of both module and RX are the last one from Frsky site LBT mode (but i tested all Flex and  FCC).
Before i was using with success the old R9M with R9mini both on this TX and on older X9D+.

The problem is that i can't register the RX to the TX. In the Register menù the radio say waiting for RX (or something similar).

Then i tested the R9M2019 module with ACCST firmware on the old X9D+ and it works fine with R9mini rx. I contacted Frsky support and after some videos and emails they sent to me their version of Opentx 2.3.9 to be flashed with DFU method. Done but without success.

Then i tried with a X10S Express of a friend running opentx 2.3.9 and it works fine. I can Register and bind R9M2019 with R9MX

 - OpenTX version 2.3.9
 - Hardware Frsky X9D+ SE 2019 - R9M2019 with firmware 1.3.0 and R9MX with firmware 1.3.2 all LBT
 - OpenTX Companion 2.3.9 windows 10

3djc, 2020-10-06:
Well we have reports of x9d 2019 working fine with r9m 2019, so that’s really strange

kilrah, 2020-10-06:
Duplicate #7983, use a nightly

uragano47812, 2020-10-06:
> Well we have reports of x9d 2019 working fine with r9m 2019, so that’s really strange. It could be a hardware defect on hb pin. Do you get module version read in radio/info screen ?

No i don't see info and is for that reason Frsky told me to install their firmware in DFU mode.

One thing i noted is that my TX is one of the first released and when i plugged in the USB charger, the module bay was with power and the led on module was ON. So Frsky sent me an autoupdate file to solve the problem and now when on charge the module is off. My friend X10S Express has the same problem but he never saw this because he doesn't use an external module.
So can it be this difference the cause of my problem?

3djc, 2020-10-06:
If you don't get version, then thats an issue. Is your module ON ?

uragano47812, 2020-10-06:
Yes module is ON and also i can flash it without problem.

3djc, 2020-10-06:
I think your radio does have a hardware defect along the HB pin path

uragano47812, 2020-10-06:
So what you suggest? Frsky didn't reply yet probably they are in holiday.

3djc, 2020-10-06:
RMA, talk to your vendor

3djc, 2020-10-06:
I'm closing this since it is almost certainly not OpenTX related

closed 2020-10-06 by 3djc

Jullibach, 2020-10-06:
I have also the SE 2019 Version of the Taranis and got also: No Information about the external R9 module. But I can bind Like normal, and fly with it every day, without any issues. In openTX 2.3.8 I got a Verison number of the External Module. So I think its a Bug in openTX and not a Hardware Defect.

mpjf01, 2020-10-07:
I'm still using 2.3.9. With an X9D+ 2019 transmitter fitted with an external R9M ACCESS module fitted I can see the module firmware version (256.15.0/1.2.0FCC). There is no module firmware information available when I fit a normal (original) R9M module.

3djc, 2020-10-07:
Yes module information are only available in access mode, thats perfectly normal

uragano47812, 2020-10-07:
> I'm still using 2.3.9. With an X9D+ 2019 transmitter fitted with an external R9M ACCESS module fitted I can see the module firmware version (256.15.0/1.2.0FCC). There is no module firmware information available when I fit a normal (original) R9M module.

Can you check if your module has led powered ON when you put the TX in charge with USB cable? Thanks

mpjf01, 2020-10-08:
No the module LED is off while charging. With both the R9M and the R9M ACCESS modules.