opentx/opentx #7996: Flysky FRM302 serial module status: not ready - OpenTX 2.3.10-OTX, x9d+, PPM works

vandusinae, 2020-10-06:
I have an issue with the AFHDS3 serial mode with Opentx 2.3.10-OTX on a x9d+ and X-Lite S (with bay adapter).
The module powers up, but can't be controlled via serial mode.

Status: not ready
Power src: unknown

If I select PPM mode everything works. I can bind the receiver and control the quad. When I switch back to serial mode, the transmitter and receiver loses connection. The blue LED on the back of the Flysky tx module flashes slowly.

3djc, 2020-10-06:
I think the module requires a specific firmware to work with OpenTX. @qba667 ?

vandusinae, 2020-10-06:
Thanks for tip! I'll try that. Cheers

qba667, 2020-10-13:
Fixed by updating TX module firmware. @vandusinae  already confirmed it is working.

closed 2020-10-13 by qba667