opentx/opentx #7997: Spectrum Analyzer not working on X9D+ with MPM (IRangeX IRX4+)

alm-nl, 2020-10-06:
I've installed OpenTX 2.3.10, but I've found an issue with the Spectrum Analyzer function used with the MPM: it does not work anymore. The screen shows the numbers, but it does not show any signal.

Transmitter: X9D+
Firmware file: opentx-x9d+-lua-en-2.3.10-otx.bin
MPM firmware: 1.3.1.69
MPM: IRangeX IRX4+

Going back to OTX 2.3.9 fixes the issue.
MPM Firmware 1.3.1.65 makes no difference.

It has been confirmed by one other person.

3djc, 2020-10-06:
Only applies to external mpm

fabioeh, 2020-10-06:
Not working on X10 t16 or xlite. Internal on t16 and X10 works.

closed 2020-10-07 by 3djc

Hobbypilots, 2021-08-12:
Same issue with me, I upgraded my T16 and Spectrum analyzer is gone. If anyone is having the old file please share.

Hobbypilots, 2021-08-12:
> I've installed OpenTX 2.3.10, but I've found an issue with the Spectrum Analyzer function used with the MPM: it does not work anymore. The screen shows the numbers, but it does not show any signal. [...]

Its challenging to find the old file, looks like Open TX was updated and Spectrum Analyzer was removed from all folders. If you have the Lua file, please share thanks

alm-nl, 2021-08-12:
This is **not** a support forum. The issue was resolved in OpenTX 2.3.11 and above. Upgrade to OpenTX 2.3.14 and latest MPM firmware (1.3.2.86 at the moment) and files and you should be ok (MPM firmware and files: https://downloads.multi-module.org/. If it still does not work after that, ask on rcgroups.com.

3djc, 2021-08-13:
@Hobbypilots there is no LUA fiel for Spektrum analyser, it is built in Otx. It requires a multi to be declared to appear

Hobbypilots, 2021-08-13:
> @Hobbypilots there is no LUA fiel for Spektrum analyser, it is built in Otx. It requires a multi to be declared to appear
I got the issue and these are my findings,
1. I upgraded to 2.3.11, missing internal RF on/ off option
2.Specktrum analyzer gone 

My fix to this problem, I found old firmware file 2.3.7 uploaded the old firmware and the issues were resolved. 

It looks like the new firmware does not allow us to switch off the internal RF and no specktrum analyzer, unless if I am missing something here.

3djc, 2021-08-13:
it just means you forgot to select internalmulti in your companion option ...

Hobbypilots, 2021-08-20:
> it just means you forgot to select internalmulti in your companion option ...

no, I have selected the internal module. The fact is it is only working with 2.3.9 anything above does not have Spektrum Analyser

commit 4db89d52 (2020-10-07): Fix #7997 Ext multi spectrum not starting