opentx/opentx #7998: Horus takes longer to boot on 2.3.10 with specific widget enabled

nmaggioni, 2020-10-06:
**Describe the bug**
Given that the active model has the [Hobby4life/GPSMap](https://github.com/Hobby4life/GPSMap) widget enabled on any page, my Horus X10S radio takes ~8s to boot (from pressing the power button to hearing the startup sound) after having updated to OTX 2.3.10; reflashing 2.3.9 brings back the shorter boot time of ~4s using the same widgets.

I have found no other widgets that cause this behavior and removing this one from the current model (or booting after having selected a model without such a widget enabled) makes the problem disappear. I'm creating an issue here since this behavior seems to be OTX-dependant rather than widget-dependant.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash OTX 2.3.9 on an Horus X10 (or presumably an equivalent radio).
2. Configure the widget linked above on any page of the currently active model.
3. Power down the radio.
4. Power up the radio, notice usual boot time.
5. Flash OTX 2.3.10.
6. Upgrade SD contents as needed.
7. Power up the radio, notice longer boot time.

**Expected behavior**
Equivalent boot times between OTX 2.3.9 and 2.3.10.

**General information**
 - OpenTX version: 2.3.10 (_opentx-x10-flexr9m-lua-noheli-en-2.3.10-otx_)
 - Hardware: FrSky X10S
 - OpenTX Companion: 2.3.10 on Manjaro Linux

3djc, 2020-10-07:
There is absolutely no changes to the boot process, so what you are saying is really strange. Please include your OTX and full  SD content

Hobby4life, 2020-10-07:
I have noticed booting time takes indeed up to 5 sec longer since v2.3.10.

3djc, 2020-10-07:
It does not without your widget, so might be worth looking at what happens on your side

Hobby4life, 2020-10-07:
I have to add, I have cleared the radio, and used a fresh SD download. (X10S) no widgets, everything blank.

robustini, 2020-10-07:
I think it's quite normal, even with Yaapu it does the same thing.

3djc, 2020-10-07:
My total boot time with 2.3.10 on x10s is 3 seconds

robustini, 2020-10-07:
Tested now, after turning on the tx I've the widget screen in exactly 7 seconds, with Yaapu the same.

3djc, 2020-10-07:
@robustini  'the same', you mean your boot time is the same between 2.3.9 and 2.3.10 ?

nmaggioni, 2020-10-07:
@3djc I can reproduce the issue with the stock SD contents, the only change being having added @Hobby4life's widget files in the SD and having enabled it on a separate page for the current model. Having seen some changes related to LUA in the changelog, I would suppose that the prolonged boot time is a consequence of something else.

I will try to bisect the latest version's commmits in the next few days, that could help pinpoint the source of the problem.

robustini, 2020-10-07:
> 
> 
> @robustini 'the same', you mean your boot time is the same between 2.3.9 and 2.3.10 ?

Yes!

3djc, 2020-10-07:
Ok so given yapuu is quite big, looks like something specific to GPSMap

Hobby4life, 2020-10-07:
As I said, I have cleared my X10S, insert a fresh SD with the latest SD contents.
There is a significant longer boot time, then v2.3.9
I have noticed it too in the nightlies, but thought it was due testing version.

3djc, 2020-10-07:
That contradicts what everyone else says in this thread

Hobby4life, 2020-10-07:
I know boot time is longer due bitmap loading of my widget.. but was never that "longer" as it is now.

Funny thing is, I dont use GPSMap myself on any of my TX at all at the moment.

robustini, 2020-10-07:
Regardless of whether the boot is longer than a few seconds I think this is not a problem, simply an observation.
3 to 7 seconds doesn't change my life.
A fast boot could be useful only in case of "emergency mode", but it depends on which aircraft you are using, automatic with failsafe or manual management.

3djc, 2020-10-07:
Emergency mode is a very fast boot mode without ui, without lua , without sounds,...

robustini, 2020-10-07:
> 
> 
> Emergency mode is a very fast boot mode without ui, without lua , without sounds,...

Exactly, problem solved.

Hobby4life, 2020-10-07:
I myself have no problems with it, but I assume that the developers would really like to know what it can be. possibility is that it may cause problems in the future. who knows?

That it happened is now clear to me after only changing firmware. there was just a slight extra delay at startup.

Could it have to do with increasing the stack size? just a thought.

3djc, 2020-10-07:
Clearly, that’s why it is still opened. @nmaggioni offered to try to narrow down the issue, let’s see where it leads to.

Hobby4life, 2020-10-07:
Let me know if testing is needed,, I have some spare horusses lying arround just for testing.

nmaggioni, 2020-10-11:
035cc6c seems to be the curlprit: the radio indeed doesn't take longer to boot, the backlight just doesn't turn on anymore until eveything else is loaded.

Before:
1. Turn on the radio.
2. The backlight turns on in ~4s and the splash screen is shown.
3. After another ~4s the startup sound is played.
4. The radio is now useable.

After:
1. Turn on the radio.
2. The backlight stays off for ~8s. If carefully looked at, the screen shows the splash screen in the meantime.
3. The backlight turns on and the startup sound is immediately played.
4. The radio is now useable.

3djc, 2020-10-11:
Ok, thanks a lot for your investigation @nmaggioni ! Closing it then

closed 2020-10-11 by 3djc

nmaggioni, 2020-10-11:
@3djc Please keep this open, I'll look into reverting to the old behavior without breaking the backlight function.

3djc, 2020-10-11:
It is done already, try with a nightly

nmaggioni, 2020-10-11:
Excellent, I didn't check them since opening this issue! Thanks a lot.

Hobby4life, 2020-10-11:
I have just tested it with the latest nightly download, but I dont see any "improvent" in a faster boot time.. still the same "longer" boot time.

Also pulled and compiled just 15mins from now.. same result.