opentx/opentx #7999: OpenTX 2.3.10 on Taranis X9D + SE 2019 - no boot after firmware flash

Ringo982, 2020-10-07:
Hello
i have just tried to upgrade my Taranis X9D + SE 2019 from opentx 2.3.9 to 2.3.10 using companion.
I have checked on my profile settings and the radio type is FrSky Taranis X9D+ 2019, i have checked only noheli and sqt5font, i have chosen my 4bit bmp image for splashscreen, downloaded the fw, installed on my taranis connected by usb in bootloader mode.
The fashing of the fw complete correctly, but as i disconnect usb cable and press exit on taranis display, the taranis turn off but the power led on the power button remain blue.
The only thing i can do is disconnect the battery, but when i try to power on the radio again it doesn't turn on... i press the power button but nothing happen.

Any idea why the installation of 2.3.10 is not working even if companion says it is successfully finished?

Thank you
Francesco

3djc, 2020-10-07:
Duplicate #7700 

Please put the firmware on your sdcard and flash from there using bootloader

closed 2020-10-07 by 3djc

Ringo982, 2020-10-07:
I have tried and doind this i can update, but the splash image don't change in any way.

I was trying using companion to see if with it i can set the image.

Can you help?

I have realizaed a 212x64 pixels BMP 4bit image...

Thank you

firozdang, 2021-01-28:
I have this same issue, it doesnt even boot now from the bootloader as well, basically my taranis is dead, doesnt boot at all now not regular not even bootloader mode.
I just upgrated the firmware to the latest OpenTX 2.3.11

3djc, 2021-01-28:
Your taranis is not dead. You need to flash it again in DFU mode

firozdang, 2021-01-28:
> Your taranis is not dead. You need to flash it again in DFU mode

But it doesnt even detect USB device anymore, it says low power USB and if I remove the battery and connect it says device malfunctioned

3djc, 2021-01-28:
Then this doesn't have much to do with the radio itself. Unplug radio battery (to be sure it is not running in an unknown state, try another usb cable and computer

firozdang, 2021-01-28:
oh wow I changed the cable it worked, now do you have any articels that I can read to debug from this point onwards?

firozdang, 2021-01-28:
I mean it only detects STM debive in DFU mode in my device manage... nothing else

3djc, 2021-01-28:
Perfect, then flash with companion. if you are using windows, it may require de DFU driver to be installed

firozdang, 2021-01-28:
C:/Program Files (x86)/OpenTX/Companion 2.3/dfu-util.exe -a 0 --dfuse-address 0x08000000:524288 --device 0483:df11  -UC:/Users/xxxxxx/AppData/Local/Temp/15000-flash-check.bin
=================================================================================
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x0483 product = 0xdf11
Opening DFU capable USB device... Cannot open device

=================================================================================

3djc, 2021-01-28:
driver issue  : https://www.hanselman.com/blog/how-to-fix-dfuutil-stm-winusb-zadig-bootloaders-and-other-firmware-flashing-issues-on-windows

firozdang, 2021-01-28:
many many many thanks, got everything working, upgraded to 2.3.11, used old EEPROM but latest firmware, you saved me money and time, thank you once again

firozdang, 2021-01-28:
FYI, this also helped: https://www.youtube.com/watch?v=hVWwInEAsa0