opentx/opentx #8004: OpenTx Companion Update Problem

ahmad-kassem, 2020-10-09:
Dear All

I am trying to update my TX16S firmware using OpenTx Companion but I am unable to download the new firmware, each time I click Download and then Download firmware i get a popup message saying: Firmware update check failed.

I uninstalled and installed again companion 2.3.10 also I tried with another computer but still facing the same problem.

I attached a screenshot of the error.

thanks in advance.

[image]

3djc, 2020-10-09:
There is an issue withour SSL certificate renewal, will be fixed shortly

3djc, 2020-10-09:
Fixed, thanks for the report

closed 2020-10-09 by 3djc