opentx/opentx #8007: OpenTX 2.3.10 - BUG Receiver ID duplicated

Ringo982, 2020-10-09:
Hello
i have a problem with my Taranis X9D + SE 2019 and OpenTX 2.3.10.
I'm using 4 quads with CRSF Micro TX v2 module.
Each quad is configured on Taranis as a specific model (quad1 model1, quad2 model2, etc), and for each model is configured on the radio the external CRSF module with a separate receiver ID (es for model1 i have set received ID 01, model2 has receiver ID se to 02, etc)
The first strange thing is that if i create a model5 and i try to set his receiver ID to 01 the radio says there are a lot of other model s that have the same receiver ID not only the first one (but it is not true!!!).
The second strange thing is that if i try to select for example model1, i power on the associate quad, the bind between MicroTX and nanoTX doesn't work. If i connect and reconnect the microTX while the quad is powered on and also the radio is powered on, for a second the bind is ok and immediately after the led on the rx become red.

The only way i have to make all working is to set all the receiver ID on all my models to 00.. using this tip every time i power on one quad it immediately bind with the microTX and the bind is stable... but the problem is that i have to be sure to have previously selected the right model on the radio  because all models now bind with all quads... and not all my configurations are the same in each model.

Have you any idea about how to solve this?

With OpenTX 2.3.9 all worked fine... each model had his receiver ID and when i had to change the model the bind happened immediately.. with the wrong model selected i had no bind...

Help please
Thank you

3djc, 2020-10-09:
While I do not have you specific radio, I tested another radio and it binds ok with any ID, and recognize ID correctly

Ringo982, 2020-10-12:
Hello, i have made some more tests. I think there is some kind of compatibility problem between OpenTX2.3.10 and Crossfire fw 3.78.
This is the situation:

- If i connect the Crossfire Micro TX v2 on my pc using TBS Agent X i can see it correctly and i can also bind correctly Nano RX receivers.
- But if i complete the binding procedure using Agent X, when i insert the Micro TX in My Taranis X9D + SE 2019 it is not possible to bind TX and RX: with radio powered on and the correct model selected, when i power on the quad the led on RX remain red. The only way to bind is to phisically disconnect the TX module on the back of the radio (with radio powered on!!!) and reconnect. But if i change model and turn back to the right model i have again the bind problem.
- The only way to let it to work properly is to bind using the radio, with the embedded crossfire scripts. If i make the bind in this way i can switch models, turn back again, power on/off radio and quads and the bind happen immediately and on the right model.

I don't know why using Agent X this is not possible, but with Open 2.3.9 fw i was able without problems to make the bind with Agent X and then transfer the Micro TX module on my Radio and use it with THAT SPECIFIC RX without problem.

Now with 2.3.10 installed on the same radio i have to rebind all my quad from radio to use them correctly.

Hope to have been clear in my explaination

Francesco

kilrah, 2020-10-12:
Well that's normal, in Agent X you can't set a receiver number so when you bind from the Agent it'll likely be 0. So it would have to be 0 in the model to work once in the radio, and binding with a non-zero number needs to be done on the radio.

closed 2020-10-29 by 3djc