opentx/opentx #8009: TX16S Trainer problem

ahmad-kassem, 2020-10-09:
Dear All

After updating my TX16S to OpenTx 2.3.10, the trainer mode: Master/Serial has disappeared, any idea Why??

3djc, 2020-10-09:
You probably have no AUX port set to trainer

ahmad-kassem, 2020-10-09:
how to solve this?

3djc, 2020-10-09:
like : set an AUX port to trainer ?

ahmad-kassem, 2020-10-09:
i found it, whey i set the serial port to sbus trainer, the option Mater/serial appears again.

3djc, 2020-10-09:
You can't set it to trainer if none of your port are set to trainer, quite logical IMHO

closed 2020-10-09 by 3djc

ahmad-kassem, 2020-10-09:
another question, now in this mode the UART is set as an SBUS signal input, Right?

3djc, 2020-10-09:
Yes, but be carefull that this is TTL level (so non inverted, 3.3v)

ahmad-kassem, 2020-10-09:
5V is not OK?

3djc, 2020-10-09:
yes, that port is 5v tolerant, but depending on your signal source, the inversion might be an issue

ahmad-kassem, 2020-10-09:
I will be using an Jumper R1+ receiver, i will weld an non-inverted sbus output from the receiver pcb and i will use it as sbus input to the UART, and i will power the receiver from the UART 5V supply, will this work?

3djc, 2020-10-09:
yes

ahmad-kassem, 2020-10-09:
another question, on the Taranis x9d the external module pins can be used as an sbus input when the trainer is set to master/sbus, is this possible in TX16S?

3djc, 2020-10-09:
no

ahmad-kassem, 2020-10-09:
ok thanks

dakhnod, 2021-05-14:
@ahmad-kassem ok, so about the tx16s:
Can I just connect some sbus receiver to one of the serial ports, set Master/Serial and opentx will read the sbus input and convert that to trainer inputs?
Also, why is it not possible to use the pins on the back as sbus input?

dakhnod, 2021-05-14:
sorry, meant to mention @3djc